renode/renode #168: Saving state crashes when simulation references STM32DMA2D or STM32LTDC

kartben, 2021-01-11:
It looks like PixelManipulationTools#PixelBlender.blender's BlenderDelegate cannot properly be serialized, hence doing the following does not work:

```
Renode, version 1.11.0.28399 (ad7ab75a-202010221522)

(monitor) s @scripts/single-node/stm32f746.resc 
(STM32F746) Save @c:/temp/xxxx
```

Stack trace:

```
Unhandled Exception: System.ArgumentNullException: Value cannot be null.
Parameter name: key
   at System.Collections.Concurrent.ConcurrentDictionary`2.GetOrAdd(TKey key, Func`2 valueFactory)
   at Antmicro.Migrant.ObjectWriter.TouchAndWriteTypeIdWithSimpleStamp(Type type)
   at Antmicro.Migrant.VersionTolerance.MethodDescriptor.Write(ObjectWriter writer)
   at Antmicro.Migrant.Utilities.IdentifiedElementsDictionary`1.TouchAndWriteId(T element)
   at HandleNewReference(ObjectWriter , Object , Int32 )
   at Antmicro.Migrant.ObjectWriter.WriteDeferredReference(Object o)
   at Write_PixelBlender(ObjectWriter , Object )
   at Antmicro.Migrant.ObjectWriter.InvokeCallbacksAndExecute(Object o, Action`1 action)
   at Antmicro.Migrant.ObjectWriter.WriteObject(Object o)
   at Antmicro.Migrant.Serializer.Serialize(Object obj, Stream stream)
   at Antmicro.Renode.Core.EmulationManager.Save(String path)
   at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid2[T0,T1](CallSite site, T0 arg0, T1 arg1)
   at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberAction(CallSite& callsite, Type binderType, Int32 knownType, LazyBinder binder, InvokeMemberName name, Boolean staticContext, Type context, String[] argNames, Object target, Object[] args)
   at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberActionCallSite(Object target, InvokeMemberName name, Object[] args, String[] tArgNames, Type tContext, Boolean tStaticContext, CallSite& callSite)
   at Dynamitey.Dynamic.InvokeMemberAction(Object target, String_OR_InvokeMemberName name, Object[] args)
   at Antmicro.Renode.UserInterface.Monitor.InvokeWithContext(InvokeContext context, MethodInfo method, Object[] parameters)
   at Antmicro.Renode.UserInterface.Monitor.ExecuteDeviceAction(Type type, String name, IEnumerable`1 p)
   at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceAction(Type device, String name, IEnumerable`1 p, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand(Token[] com, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ParseTokens(IEnumerable`1 tokensToParse, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.Parse(String cmd, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.HandleCommand(String cmd, ICommandInteraction ci)
   at AntShell.Shell.HandleCommand(String cmd, ICommandInteraction ic)
   at AntShell.CommandLine.HandleControlSequence(ControlSequence seq)
   at AntShell.Terminal.NavigableTerminalEmulator.HandleInput(Object input)
   at AntShell.Terminal.NavigableTerminalEmulator.Run(Boolean stopOnError)
   at AntShell.Shell.Start(Boolean stopOnError)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart(Object obj)
```

mateusz-holenko, 2021-01-18:
Thanks for reporting the issue, @kartben !

98814a7d1579bf8002a707f6cc5fbe3eba7cfa71 should solve problems with the serialization of `STM32DMA2D` and `STM32LTDC` models.

When fixing this we noticed, however, an issue with serialization of UART devices (not strictly STM32-related) on Windows (it seems to work fine on Linux). We are investigating it.

mateusz-holenko, 2021-02-01:
@kartben: https://github.com/renode/renode/commit/57cc3b736e8d99a6deeceb769d7553f1daf1ab8d should solve the problem of UART devices serialization on Windows. 

Could you verify if it works for you now?

kartben, 2021-02-03:
@mateusz-holenko perfect, it is working now -- thanks!

closed 2021-02-03 by kartben