renode/renode #172: STM32DMA2D - Memory-to-memory with PFC should support alpha value replacement

kartben, 2021-01-19:
As per the reference manual for memory-to-memory with pixel format conversion: 

> In parallel to the color conversion process, the alpha value can be added or changed depending on the value programmed in the `DMA2D_FGPFCCR` register. If the original image does not have an alpha channel, a default alpha value of `0xFF` is automatically added to obtain a fully opaque pixel. The alpha value can be modified according to the `AM[1:0]` bits of the `DMA2D_FGPFCCR` register

I've implemented a fix for this and should be submitting a PR (alongside other fixes to DMA2D and LTDC!) shortly.

kartben, 2021-01-30:
Addressed in PR renode/renode-infrastructure#31

PiotrZierhoffer, 2021-05-18:
This is actually merged :)

closed 2021-05-18 by PiotrZierhoffer