renode/renode #173: Documentation for config file

Barabas5532, 2021-01-20:
I can't find any documentation for the config file. Does this exist?

I was able to change the terminal application by changing the `terminal = Termsharp`
 line to `terminal = gnome-terminal`. It seems to work, but I'm not sure if I'm breaking anything by doing this. Main motivation for this was to get rid of the white on black colourscheme of the default terminal.
 
 Are there any hidden options?
 
 Here is my full config file for reference:
 
```
[general]
terminal = gnome-terminal
collapse-repeated-log-entries = True
log-history-limit = 1000
use-synchronous-logging = False
always-log-machine-name = False
history-path = /home/me/.config/renode/history
[monitor]
consume-exceptions-from-command = True
break-script-on-exception = True
number-format = Hexadecimal
[plugins]
enabled-plugins = 
[termsharp]
append-CR-to-LF = True
font-face = Roboto Mono
font-size = 10
[file-system]
use-cow = False
[file-fetcher]
calculate-checksum = True
```

Ristovski, 2021-01-26:
You should be able to find all of them by searching for `ConfigurationManager` in the [renode-infrastructure](https://github.com/renode/renode-infrastructure/search?q=ConfigurationManager) repo.  
Sadly, they are not defined in a single place.

PiotrZierhoffer, 2021-01-26:
The available options are `Termsharp`, `GnomeTerminal` and `TerminalApp` for macOS. `gnome-terminal` silently fell back to `XTerm`, I believe. We need to make it more verbose.

Termsharp does include several features (like displaying images) and is by far the most tested front-end we have. It's also cross-platform, so works on Windows, macOS and Linux.

The config file documentation is something that we don't yet have, but it's a valid point to add it.

PiotrZierhoffer, 2021-01-26:
@Ristovski the rationale behind scattered config options is that it allows to easily add new ones from plugins and user-specific code. It could generate the docs nevertheless.

ryan-mcclue, 2021-10-12:
@Barabas5532 How do you load this config file? What is it named and where should it be?

Ristovski, 2021-10-12:
@ryan-mcclue `~/.config/renode/config` on Linux. It gets the path like this, which might give you an idea where it looks on Windows/Mac:

```C#
UserDirectoryPath = Path.Combine(Environment.GetFolderPath(Environment.SpecialFolder.ApplicationData), "renode");
```