renode/renode #174: Provide docker file/image to build Renode

Viatorus, 2021-01-21:
As the title says - it would to be great if you could publish a docker file/image to build renode, including the linux portable version.

If I try it myself like described [here](https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html)
I don't get a working GUI.

If I try to build the portable version and try to run it, I get the following exception:

```
Unhandled Exception:
System.TypeInitializationException: The type initializer for 'Antmicro.Renode.Logging.ConsoleBackend' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.Console' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.ConsoleDriver' threw an exception. ---> System.TypeInitializationException: The type initializer for 'Sys' threw an exception. ---> System.EntryPointNotFoundException: SystemNative_LChflagsCanSetHiddenFlag assembly:<unknown assembly> type:<unknown type> member:(null)
  at (wrapper managed-to-native) Interop+Sys.LChflagsCanSetHiddenFlag()
  at Interop+Sys..cctor () [0x00000] in <d13c8b563008422a8c5aaec0a74089cc>:0 
   --- End of inner exception stack trace ---
  at System.IO.FileSystem.FileExists (System.ReadOnlySpan`1[T] fullPath, System.Int32 fileType, Interop+ErrorInfo& errorInfo) [0x00007] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.IO.FileSystem.DirectoryExists (System.ReadOnlySpan`1[T] fullPath, Interop+ErrorInfo& errorInfo) [0x00000] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.IO.FileSystem.DirectoryExists (System.ReadOnlySpan`1[T] fullPath) [0x00000] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.IO.Directory.Exists (System.String path) [0x0001e] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.TermInfoDriver.SearchTerminfo (System.String term) [0x00044] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.TermInfoDriver..ctor (System.String term) [0x0004b] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.ConsoleDriver.CreateTermInfoDriver (System.String term) [0x00000] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.ConsoleDriver..cctor () [0x0004d] in <d13c8b563008422a8c5aaec0a74089cc>:0 
   --- End of inner exception stack trace ---
  at System.Console.SetupStreams (System.Text.Encoding inputEncoding, System.Text.Encoding outputEncoding) [0x00007] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.Console..cctor () [0x0007d] in <d13c8b563008422a8c5aaec0a74089cc>:0 
   --- End of inner exception stack trace ---
  at Antmicro.Renode.Logging.ConsoleBackend..ctor () [0x00000] in <1db6580045684397b7c47657fbae8dfa>:0 
  at Antmicro.Renode.Logging.ConsoleBackend..cctor () [0x00000] in <1db6580045684397b7c47657fbae8dfa>:0 
   --- End of inner exception stack trace ---
  at Antmicro.Renode.Program.ConfigureEnvironment () [0x00000] in <db166c9be8be4815b35720b261227f7b>:0 
  at Antmicro.Renode.Program.Main (System.String[] args) [0x00035] in <db166c9be8be4815b35720b261227f7b>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.TypeInitializationException: The type initializer for 'Antmicro.Renode.Logging.ConsoleBackend' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.Console' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.ConsoleDriver' threw an exception. ---> System.TypeInitializationException: The type initializer for 'Sys' threw an exception. ---> System.EntryPointNotFoundException: SystemNative_LChflagsCanSetHiddenFlag assembly:<unknown assembly> type:<unknown type> member:(null)
  at (wrapper managed-to-native) Interop+Sys.LChflagsCanSetHiddenFlag()
  at Interop+Sys..cctor () [0x00000] in <d13c8b563008422a8c5aaec0a74089cc>:0 
   --- End of inner exception stack trace ---
  at System.IO.FileSystem.FileExists (System.ReadOnlySpan`1[T] fullPath, System.Int32 fileType, Interop+ErrorInfo& errorInfo) [0x00007] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.IO.FileSystem.DirectoryExists (System.ReadOnlySpan`1[T] fullPath, Interop+ErrorInfo& errorInfo) [0x00000] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.IO.FileSystem.DirectoryExists (System.ReadOnlySpan`1[T] fullPath) [0x00000] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.IO.Directory.Exists (System.String path) [0x0001e] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.TermInfoDriver.SearchTerminfo (System.String term) [0x00044] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.TermInfoDriver..ctor (System.String term) [0x0004b] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.ConsoleDriver.CreateTermInfoDriver (System.String term) [0x00000] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.ConsoleDriver..cctor () [0x0004d] in <d13c8b563008422a8c5aaec0a74089cc>:0 
   --- End of inner exception stack trace ---
  at System.Console.SetupStreams (System.Text.Encoding inputEncoding, System.Text.Encoding outputEncoding) [0x00007] in <d13c8b563008422a8c5aaec0a74089cc>:0 
  at System.Console..cctor () [0x0007d] in <d13c8b563008422a8c5aaec0a74089cc>:0 
   --- End of inner exception stack trace ---
  at Antmicro.Renode.Logging.ConsoleBackend..ctor () [0x00000] in <1db6580045684397b7c47657fbae8dfa>:0 
  at Antmicro.Renode.Logging.ConsoleBackend..cctor () [0x00000] in <1db6580045684397b7c47657fbae8dfa>:0 
   --- End of inner exception stack trace ---
  at Antmicro.Renode.Program.ConfigureEnvironment () [0x00000] in <db166c9be8be4815b35720b261227f7b>:0 
  at Antmicro.Renode.Program.Main (System.String[] args) [0x00035] in <db166c9be8be4815b35720b261227f7b>:0
```

PiotrZierhoffer, 2021-01-26:
Hi @Viatorus 

We used to have such a docker image, but we decided to make it smaller. The last renode-docker commit that had all the build-time dependencies was https://github.com/renode/renode-docker/commit/9dc5fff64e6e8a13d9459aca9fccfb6f11045f18

We will probably create a renode-dev image, but we don't have specific timelines.

Having said that, this was tested on regular package builds, not the portable one.

Regarding the error - do you try to build and run on the same system? If not, what distribution do you build and what distribution do you run on?

PiotrZierhoffer, 2021-01-26:
To have a GUI working from Docker you need to build pass the XAuthority file: https://github.com/renode/renode#running-renode-in-a-docker-container