renode/renode #175: SysTick IRQ interferes with single-step debugging & oscillator init issues on STM32F4 Discovery

grifcj, 2021-01-21:
The SysTick IRQ continuously fires after being enabled, interfering with source level debugging. It is as though the core clock never halts with gdb attached like it typically would with a debugger attached to a hardware target. Is this just how things work or am I missing a setting controlling how renode interacts with gdb, timers, and virtual time?

I used the STCubeMx to generate a makefile project for the STM32F407 Discovery board.
[image]

System Info:
```
Ubuntu 20.04.1 LTS
Renode, version 1.11.0.24832 (ad7ab75a-202010221310)
```

The renode script should be visible in top left pane. The renode log in bottom pane. gdb is on right screen 

[image]

I was debugging an issue with oscillator config at boot and think I'm narrowing down on it via breakpoints, noisy logs, and function name logging (pretty sure it's unimplemented flash controller register at fault). Just curious if there might be something to make single-stepping more functional with gdb, like ensure stepping only steps time forward the amount necessary to execute next instructions in thread of control or better synchronizes gdb with the passing of virtual time. I do not have a hardware target to verify my recollections, but they are that if I was using a typical ICD, i'd be able to single-step my way to the offending problem spot in early boot because the debugger would halt the core-clock after every step, prevention expiration of SysTick. The interrupt wouldn't continuously trigger between steps.

My guess is that there isn't really an issue here to reproduce, but if there is and you would like to replicate the build, I can create a repo for it.

kartben, 2021-01-21:
+1, I'm observing a similar behaviour. Also, virtual time seems to keep going even when simulation is paused on a breakpoint, which is probably related?
cc @PiotrZierhoffer

PiotrZierhoffer, 2021-01-21:
This actually not strictly related. We have solved this issue in RISC-V but we need to devise a cross-architectural solution

IvanVnucec, 2021-02-09:
Any info how to solve this?

IvanVnucec, 2021-02-09:
Also, I can confirm that the stm32f103 is also affected by thin bug.

PiotrZierhoffer, 2021-02-10:
We are implementing a fix that will stop the virtual time flow. This will mitigate the issue, i.e. interrupts will happen, but significantly less often.

Blocking interrupts while debugging is the next step we'll be working on. 

Meanwhile, to mitigate the interrupts from systick, the only way would be to actually mask them, or change priorities, on the nvic level.

IvanVnucec, 2021-02-10:
> We are implementing a fix that will stop the virtual time flow. This will mitigate the issue, i.e. interrupts will happen, but significantly less often.
>
> Blocking interrupts while debugging is the [...]

Can you give more info about masking and priority changing? 
I'm developing RTOS for Cortex-M3 CPU and the SysTick for me is important thing to have.

IvanVnucec, 2021-04-17:
@PiotrZierhoffer What is the status with this issue?

mateusz-holenko, 2021-05-13:
In https://github.com/renode/renode/commit/48e516dcb6cc6c011e680491b657d704b2a6a6e1 we merged a feature called "Debug Mode" that disables interrupts when stepping in GDB. 

Add the following line to your script (or enter it directly in Monitor) in order to enable it:
```
sysbus.cpu ShouldEnterDebugMode true
```

With this you should not jump to the SysTick IRQ handler anymore as long as you are *stepping* through the code.

Let me know if this solves the problem.

grifcj, 2021-05-14:
This resolved the issue.  

Also checked that, if for some reason, you did want to break on the SysTick interrupt, you can place a breakpoint there and everything works as expected. Same with other interrupts and of course normal code.

closed 2021-05-14 by grifcj