renode/renode #177: ARMv7m MPU (PMSAv7) support

rgujju, 2021-01-31:
Does renode support the ARMv7m MPU (PMSAv7)? I can see only MPUType implemented currently.
If it does, how do i enable it? 
If not, then how do i add support for it? Not only for handling the read/write to the MPU registers but implementing an actual MPU support so that it provides fault/error messages when the code tries to execute from a blocked region.
I reckon this support would need to be added in tlib, right?

PiotrZierhoffer, 2021-02-02:
The implementation would cover both tlib and renode-infrastructure (for the register API).

This MPU is not yet on master, but we do have partial support developed internally and tested with Zephyr. It will take some time to release it though, as it's a significant part of code affecting many platforms.

PiotrZierhoffer, 2026-10-05:
This is done for a very long time, closing

closed 2026-10-05 by PiotrZierhoffer