renode/renode #178: Symantec crashes Renode on Windows

grifcj, 2021-01-31:
Latest Renode release doesn't run on Windows with Symantec.

From Windows Event Viewer:
```
Faulting application name: Renode.exe, version: 1.11.0.28399, time stamp: 0x5f918d3f
Faulting module name: sysfer.dll, version: 14.3.227.0, time stamp: 0x5e213289
Exception code: 0xc0000005
Fault offset: 0x0000000000093da9
Faulting process id: 0xe90
Faulting application start time: 0x01d6f8167ec6482d
Faulting application path: C:\Program Files\Renode\bin\Renode.exe
Faulting module path: C:\WINDOWS\System32\sysfer.dll
Report Id: 9d5d079a-1462-479b-a308-ebb67c392270
Faulting package full name: 
Faulting package-relative application ID: 
```

Actually, it's a bit of a guess that it's Symantec at fault here based on some googling about sysfer.dll
[Some reference to it in broadcom](https://knowledge.broadcom.com/external/article/181736/how-to-create-an-application-control-exc.html)
[StackOverflow referencing exception before main](https://stackoverflow.com/questions/33741696/vs2015-several-projects-throwing-exceptions-before-main)

I will work with my IT to get an exception in place for Renode (which will confirm Symantec is the issue). Whether something can be done with Renode (or its runtime) to address this, suppose that will take some more digging.

PiotrZierhoffer, 2021-02-02:
Thanks for the report.

This is quite surprising, indeed. I am not yet sure how to address this, but we will try to see if there's an easy workaround,

grifcj, 2021-11-25:
Closing this just to clean up my issues. I no longer work with the system in question and this seemed pretty obvious to be a Symantec problem that Renode wouldn't be able to do much about.

closed 2021-11-25 by grifcj