renode/renode #179: GDB-server fails after sysbus.cpu Reset

RamiRosenbaum, 2021-02-01:
Hi,
I'm running Renode version 1.11.0.28399 on Windows 10.
I'm trying to re-debug my application by resetting the debugger, after hitting a breakpoint, but am facing serious stability issues.
Sometimes my breakpoint (E.G.: `(gdb) b main`) will hit twice, after the initial run and the following reset, but never a third time, and many times not even the second time.
My scenario:
1. Setup the Renode machine
2. Start the Renode GDB-server (`machine StartGdbServer 4242`)
3. Connect via GDB-client (`arm-none-eabi-gdb -ex "tar ext :4242" .\my_app.elf`) and set a breakpoint (`(gdb) b main`)
4. Run :
`(gdb) monitor pause`
`(gdb) monitor sysbus.cpu Reset`
`(gdb) monitor start`
`(gdb) c`

Repeating step 4 should restart the emulated applicaction and hit the breakpoint, but it happens not more than twice.
**Running this scenario on Linux is stable...**
Thanks.

PiotrZierhoffer, 2021-02-02:
Hi 

Thanks for the report. 

I believe you can achieve you goal by resetting the whole machine: 
`(gdb) monitor machine Reset`

This resets the state of all peripherals, including the CPU.
It does require you to have a proper `reset` macro included in your script, for the application to be loaded.

In most cases it will be very simple, like here: https://github.com/renode/renode/blob/master/scripts/single-node/sam_e70.resc#L17

RamiRosenbaum, 2021-02-03:
Hi, thanks for the quick reply.
Unfortunately, this is not the required behavior for my testing.
I'm simulating something similar to a journaling flash file system, so restarting the machine (**and erasing the flash on the way**) is exactly what I want to avoid!
I want to simulate a _naïve_ RESET of the board, where the flash image keeps its history.
Thanks.

PiotrZierhoffer, 2021-02-03:
So the problem is that `sysbus.cpu Reset` actually does not set the PC to the entry point. This is done by `LoadELF`, which you don't run, because you reset the CPU only.

Can you try the following?
```
(gdb) monitor p
(gdb) mon sysbus.cpu Reset
(gdb) mon sysbus.cpu PC 0xDEAFBEEF (put in whatever the entry point for your application is)
(gdb) mon s
(gdb) c
```

I noticed an issue when you try to place your breakpoint exactly at this initial PC, but I believe that's not exactly what you're tackling here.

PiotrZierhoffer, 2021-02-03:
> try to place your breakpoint exactly at this initial PC

I actually believe this is GDB-related, as you're already at the entry point when you connect.

RamiRosenbaum, 2021-02-03:
Using the following magic-mixture worked for 2 resets (3 runs), but very unstable; after many attempts.

PiotrZierhoffer, 2021-02-03:
This is very strange. Is `main` the entry point of your ELF? Or is it, s I assumed, just one of the functions?

PiotrZierhoffer, 2021-05-18:
Closing this one as we got not reply. @RamiRosenbaum feel free to reopen if you have any more details.

closed 2021-05-18 by PiotrZierhoffer