renode/renode #180: additional-prerequisites

IvanVnucec, 2021-02-02:
I had installed Renode on Ubuntu by following Installation documentation.
I've installed everything except Robot testing framework. 
I ran into unknown because the documentation states that I should run pip to install all the dependencies written in `tools/requirements.txt` but in folder where I've unpacked version 1.11 there is no folder `tools`.
How do I install the Robot framework.

https://github.com/renode/renode/#additional-prerequisites-for-robot-framework-testingInstallation

IvanVnucec, 2021-02-02:
Ok, I've installed Robot framework by running `pip install robotframework` like it is written here: https://pypi.org/project/robotframework/#installation.

Maybe the Renode installation documentation should be updated. Also I was not sure about Ubuntu version: it is written `On Ubuntu 16.04, you can install the remaining dependencies with the following command:` but what if I have Ubuntu version 20.04?
https://github.com/renode/renode/#other-dependencies-linux-only

PiotrZierhoffer, 2021-02-02:
Thanks for the report

The file is in `tests/requirements,txt` if you install from the packages. We'll clean it up shortly. For robotframework please use the version mentioned in that file.

I believe Ubuntu 20.04 will have the very same command - but we will double check and update the readme accordingly

PiotrZierhoffer, 2021-02-03:
I verified that Ubuntu 20.04 works the same way, so I bumped it in README. I consider this closed, but if you have any other issues feel free to reopen or open new issues

closed 2021-02-03 by PiotrZierhoffer

IvanVnucec, 2021-02-03:
ok, thank you.