renode/renode #181: Unable to find provided CSS file renode\tests\../lib/resources/styles/robot.css

kaizoku-oh, 2021-02-08:
When trying to run test with the following command:
`python -u D:/renode/tests/run_tests.py test/test-button.robot -r test_results --variable PATH:$PWD`

I get the following error:
`Unable to find provided CSS file: D:\renode\tests\../lib/resources/styles/robot.css`

And I can't find the robot.css anywhere in the renode repo.
Am I missing something?

PiotrZierhoffer, 2021-02-08:
Thanks for the report.

Do you run Renode from a cloned repo? Or is it a Windows package?
Did you run build.sh? One of the steps it performs is downloading the repo under `lib/resources`: https://github.com/renode/renode-resources.

It contains the stylesheet and is usually fetched during the build process.

kaizoku-oh, 2021-02-09:
Thanks for answering.
I installed renode from the msi windows package.
And I also cloned the repo (I thought it was necessary to be able to use the robot framework).
As for renode-resourses: how can I install them on windows 10?

PiotrZierhoffer, 2021-02-09:
It should be in the package, no need to install anything additional. I'll verify that asap

PiotrZierhoffer, 2021-02-09:
If you have installed in the default path, the file should be in `c:\Program Files\Renode\tests\robot.css`. You can provide the path with `python -u D:/renode/tests/run_tests.py --css-file c:/Program\ Files/Renode/tests/robot.css`

This will be addressed in the next package release, thanks for spotting this

kaizoku-oh, 2021-02-09:
Thank you for your support, here is the full command that worked for me in windows 10 powershell in case someone needs this:
`python -u D:/renode/tests/run_tests.py --css-file C:/Program` Files/Renode/tests/robot.css --robot-framework-remote-server-full-directory C:/Program` Files/Renode/bin test/test-button.robot -r test_results --variable PATH:$PWD`

closed 2021-02-09 by kaizoku-oh