renode/renode #182: GDB not showing correct LiteX_CPUTimer value

dayjaby, 2021-02-09:
Hi,

In renode I got the following CPUTimer running:
```
(machine-0) sysbus.cpu_timer TimerValue
0x0000000173463BF0
(machine-0) peripherals
Available peripherals:
  sysbus (SystemBus)
  │   
  ├──  (Silencer)
  │     <0xF0003000, 0xF00031FF>
  │       
  ├──  (Silencer)
  │     <0xF0003800, 0xF00039FF>
  │       
  ├── cpu (VexRiscv)
  │     Slot: 0
  │       
  ├── cpu_timer (LiteX_CPUTimer)
  │     <0xF0000800, 0xF00008FF>
  │ 
```

However, when I'm reading values at this address via GDB, I get just zeroes:
```
(gdb) x/z 0xF0000800
0xf0000800:	0x00000000
(gdb) x/z 0xF0000804
0xf0000804:	0x00000000
(gdb) x/z 0xF0000808
0xf0000808:	0x00000000
(gdb) x/z 0xF000080c
0xf000080c:	0x00000000
(gdb) x/z 0xF0000810
0xf0000810:	0x00000000
```

Am I missing a step or is GDB unable to read such values? Even when setting a breakpoint into zephyr's mtime function, it reads only zeroes.

dayjaby, 2021-02-09:
Maybe these warnings are also relevant:
```
09:42:39.9867 [WARNING] cpu_timer: Unhandled write to offset 0xC. Unhandled bits: [8-10, 13] when writing value 0x2710.
09:42:40.0066 [WARNING] cpu_timer: Unhandled write to offset 0xC. Unhandled bits: [8-9, 11-12, 16-19] when writing value 0xF1B30.
```

PiotrZierhoffer, 2021-02-15:
A quick look at the model shows that you need to latch the value in GDB before reading it out. So you need to write a non-zero value to 0xF0000800 and then you can read out the registers starting at 0xF0000804.

I am not 100% sure how well does this align with the timer's RTL, but this way you should be able to read correct values.

dayjaby, 2021-02-25:
That makes sense. Thank you!

closed 2021-02-25 by dayjaby