renode/renode #184: SystemBus: consider caching "WhatIsAt()" (and potentially all other LINQ queries)

kartben, 2021-02-15:
On a simple Mbed OS app for NRF52840, I've been able to get a 3x simulation perf improvement by caching the results of `SystemBus.WhatIsAt()`. 
Any thoughts on whether this is worth a PR? I can't think of any issues associated to the change granted:
- the cache should hopefully not grow like crazy, as I would assume peripherals tend to use a reasonably small address space?
- the cache can be aggressively cleared any time a peripheral is added/removed, although a smarter algorithm can probably be implemented too.

FWIW it might be interesting to look at potentially implementing similar caches for all the places in SystemBus where `peripherals.Peripherals.Where(...)` is called, but it might be more difficult to do it right for mapped memory etc.

PiotrZierhoffer, 2021-02-15:
I believe you actually spotted a bug that affected the performance of writes in Renode in general.

This is a very good catch! We have found this in reads before, but somehow writes remained not optimal.

Caching is actually not needed here. `WhatPeripheralIsAt` et al are not really supposed to be called by the running emulation. They are mostly there for the UI/CLI use.

My quick check shows that `peripherals.Peripherals.Where(...)` is not used anywhere apart of initialization or platform creation, so I don't consider that to be a priority right now.

I'm running tests on a fairly simple fix and will push it as soon as we're green.

PiotrZierhoffer, 2021-02-16:
@kartben can you please verify if the changes I just pushed help?

kartben, 2021-02-16:
They do help! Thanks!

closed 2021-02-16 by pgielda