renode/renode #185: Does renode support outputting JUnit XML format?

mithro, 2021-02-15:
JUnit XML is a format used by many test suite systems and highly supported by many CI systems. Does Renode support outputting this format?

https://www.ibm.com/support/knowledgecenter/SSQ2R2_9.1.1/com.ibm.rsar.analysis.codereview.cobol.doc/topics/cac_useresults_junit.html

PiotrZierhoffer, 2021-02-15:
While Renode does not support it by default, the Robot Framework has some xUnit support. It's not super robust but I used it with success uploading test results to some external. It's also, obviously, less detailed.

With the Robot Framework installed (and this is required to run tests anyway) it's very easy to convert the results (usually stored in robot_output.xml) to xUnit:

```
rebot --xunit xunit_results.xml robot_output.xml
```

AntonKrug, 2021-02-16:
'rebot'? is that a typo?

PiotrZierhoffer, 2021-02-16:
No, it's a Robot Framework tool for output processing :)

closed 2021-05-18 by PiotrZierhoffer