renode/renode #186: Http server does not reply the request

omidfatemi, 2021-02-16:
Hello, I have renode on my virual Machine with Ubuntu 20.04.2 LTS, I am running a code to run an http server on quark c1000 and the code is as follows, 
`
emulation CreateSwitch "switch"
emulation CreateTap "tap0" "tap"
connector Connect host.tap switch
$bin ?= @https://dl.antmicro.com/projects/renode/http_server.elf-s_831660-df4e7a424882a5eb4883dddb3988971760732f78
$name?="Server"
mach create $name
machine LoadPlatformDescription @platforms/boards/arduino_101-shield.repl
$boot ?= @https://dl.antmicro.com/projects/renode/quark_se_rom.bin-s_8192-b688c8b9380014d7deb2bd42dc218fc5ee8d1abf
using sysbus
showAnalyzer uartB
macro reset
"""
    # Set the WAIT_FOR_JTAG GPIO pin. The bootloader stalls if it's set to low.
    gpio OnGPIO 15 true

    # Load the bootloader. The address has to be specified as the binary files do not contain
    # any addressing information.
    sysbus LoadBinary $boot 0xffffe000
    # The ELF files already contain information on where to load each part. They also provide symbols, which
    # can be used for execution tracing.
    sysbus LoadELF $bin

    # By default, the initial PC is set to the ELF entry point. Since the bootloader has to be started before the target application,
    # the PC needs to be set manually to the correct value.
    sysbus.cpu PC 0xfffffff0
"""
runMacro $reset
connector Connect spi1.ethernet switch
start`
the ping from host to server is ok, but when I request index.html with 
`wget 192.0.2.2:80/index.html -O index.html`
there's no reply from the server.
your help is much appreciated.

mateusz-holenko, 2021-02-17:
Hi @omidfatemi, thanks for reporting the issue!

The robot test for this particular demo (https://github.com/renode/renode/blob/master/tests/platforms/QuarkC1000/QuarkC1000.robot#L149) seems to work fine on my local machine (provided that I have the `tap0` interface configured properly).

I checked it with `ncat` and it works as well:
```
echo "GET /index.html HTTP/1.0" | ncat 192.0.0.2 80
```

I was, however, able to reproduce the problem when using `wget` or `curl`.
I suspect it might have something to do with the request content generated by those tools, particularly with the length of header fields.

The following is the maximal request the HTTP server was able to handle correctly - extending the custom header with just one character causes the server not to respond:
```
GET /index.html HTTP/1.1
custom-header: 12345678901234567890123456789012
```

I suggest you could try building and running a current Zephyr binary - perhaps there was some software bug that is now fixed on master?

omidfatemi, 2021-02-17:
> Hi @omidfatemi, thanks for reporting the issue!
>
> The robot test for this particular demo (https://github.com/renode/renode/blob/master/tests/platforms/QuarkC1000/QuarkC1000.robot#L149) seems to work [...]

Thanks a lot Mateusz  for your reply, I also checked with ncat, but it didnt work, also I attached the log and  a screenshot of the log to this message since it is different from the video tutorial, 
[output.txt](https://github.com/renode/renode/files/5995847/output.txt)
[image]
I could not  find any issue previously reported related to this case.

mateusz-holenko, 2021-05-18:
Could you check if the robot test works on your setup?

You can run it using the following command:
```
renode-test @tests/platforms/QuarkC1000/QuarkC1000.robot -f "*Should Serve Page Using Tap"
```
(NOTE: if you built Renode from sources the script will be names `test.sh` instead of `renode-test`)

omidfatemi, 2021-05-19:
> renode-test @tests/platforms/QuarkC1000/QuarkC1000.robot -f

hello, I ran the code, I got 
[http_ctx_get:296] Free ctx found, index: 0
on the uart, but the index.html file was not recieved:
HTTP request sent, awaiting response...

mateusz-holenko, 2021-05-21:
Could you share the test output files: `log.html`, `report.html` and `robot_output.xml`?

omidfatemi, 2021-05-22:
sure, I output the log in log.txt which is attached
[log.txt](https://github.com/renode/renode/files/6527203/log.txt)

mateusz-holenko, 2021-05-27:
What I meant by `log.html`, `report.html` and `robot_output.xml` were files generated by the robot testing framework. 
You can see find their paths at the end of the test run message:
```
$ ./test.sh tests/platforms/QuarkC1000/QuarkC1000.robot -f "*Should Serve Webpage Using Tap"
Testing fixture: *Should Serve Webpage Using Tap
Preparing suites
Started Renode instance on port 9999; pid 710478
Starting suites
Running tests/platforms/QuarkC1000/QuarkC1000.robot
+++++ Starting test 'QuarkC1000.Should Serve Webpage Using Tap'
+++++ Finished test 'QuarkC1000.Should Serve Webpage Using Tap' in 2.09 seconds with status OK
Cleaning up suites
Closing Renode pid 710478
Aggregating all robot results
Output:  /home/antmicro/renode/output/tests/robot_output.xml
Log:     /home/antmicro/renode/output/tests/log.html
Report:  /home/antmicro/renode/output/tests/report.html
Tests finished successfully :)
```

ajelinski, 2021-08-24:
Hi @omidfatemi!
This issue got back to us as it seems the HTTP request sent in the `QuarkC1000.Should Serve Webpage Using Tap` test somehow grew beyond Zephyr's frame size limit found in `http_server`.
It seems to be because of some changes in `requests` Python library, which is used to make that request.
The limit seems to be 128B and it indeed prevents successful usage of `wget` and `curl` with default headers.

@mateusz-holenko's `ncat` command should work for you but please pay attention to the IP.
It should be 192.0.**2**.2 so it might be the case why it didn't work for you (unless you have noticed and fixed that).
Curl should also work if the default headers are removed: `curl -H 'User-Agent:' -H 'Accept:' -H 'Host:' 192.0.2.2/index.html`.

BTW. We're going to replace `http_server` in the test soon to this one that can handle much larger frames (even 500B): https://dl.antmicro.com/projects/renode/http_server-s_819052-7ddc5a39c953a6330ca4971bd99bda1ea2a5beca
Using it as `$bin` should be helpful for you too but please remember to use port 8000 with it (the old one uses port 80).