renode/renode #189: CC1200 not working when compiling for firefly in Contiki-NG

tinstructor, 2021-03-12:
The issue I'm having is that when I try to simulate three firefly nodes running [the contiki-ng rpl-udp example](https://github.com/contiki-ng/contiki-ng/tree/develop/examples/rpl-udp), the CC1200 radio does not seem to be simulated correctly and hence no traffic is exchanged. 

To reproduce this issue you must first create a script called `firefly.resc` in the [rpl-udp directory](https://github.com/tinstructor/contiki-ng/tree/develop/examples/rpl-udp) and copy the following contents into it:

```
:name: FIREFLY
:description: This script runs the twofaced node example from Contiki-NG on Zolertia Firefly.

$name?="firefly"
$bin?=@build/zoul/firefly/udp-client.elf

using sysbus
mach create $name

machine LoadPlatformDescription @platforms/boards/zolertia-firefly.repl
machine PyDevFromFile @scripts/pydev/rolling-bit.py 0x400D2004 0x4 True "sysctrl"

$id = `next_value 1`

macro reset
"""
    #set node address. 0x00 0x12 0x4B is TI OUI
    sysbus WriteDoubleWord 0x00280028 $id
    sysbus WriteDoubleWord 0x0028002C 0x00
    sysbus WriteDoubleWord 0x00280030 0xAB
    sysbus WriteDoubleWord 0x00280034 0x89
    sysbus WriteDoubleWord 0x00280038 0x00
    sysbus WriteDoubleWord 0x0028003C 0x4B
    sysbus WriteDoubleWord 0x00280040 0x12
    sysbus WriteDoubleWord 0x00280044 0x00

    sysbus LoadBinary @https://dl.antmicro.com/projects/renode/cc2538_rom_dump.bin-s_524288-0c196cdc21b5397f82e0ff42b206d1cc4b6d7522 0x0
    sysbus LoadELF $bin
"""

runMacro $reset

showAnalyzer sysbus.uart0
```

Next, create a script called `cc1200-test.resc` in the same directory with the following contents:

```
:name: firefly range medium 3 nodes

######################################################################
#
# Sample script creating 1 root node and 2 non-root nodes using a
# range-based medium function - transmission is possible if nodes are
# in range. Nodes are positioned in a 3D space (x,y,z),
# at a distance of 10 (units) from each other.
#
######################################################################

emulation CreateWirelessMedium "wireless"

# The radio is using a range-based medium (with the `Range` set to `11`)
# If not set, the default SimpleMediumFunction will be used (where range is not relevant)
wireless SetRangeWirelessFunction 11

######################### ROOT NODE - begin #########################
$bin=@build/zoul/firefly/udp-server.elf
$name="root"
i @firefly.resc
connector Connect radio wireless
wireless SetPosition radio 0 0 0
mach clear
########################## ROOT NODE - end ##########################

######################### NON-ROOT NODE - begin #########################
$bin=@build/zoul/firefly/udp-client.elf
$name="node-1"
i @firefly.resc
connector Connect radio wireless
wireless SetPosition radio 10 0 0
mach clear
########################## NON-ROOT NODE - end ##########################

######################### NON-ROOT NODE - begin #########################
$name="node-2"
i @firefly.resc
connector Connect radio wireless
wireless SetPosition radio 0 10 0
mach clear
########################## NON-ROOT NODE - end ##########################
```

Finally, add a `project-conf.h` header file in the same directory and add to it the following:

```c
#ifndef PROJECT_CONF_H_
#define PROJECT_CONF_H_

#define NETSTACK_CONF_RADIO             cc1200_driver
#define CC1200_CONF_RF_CFG              cc1200_802154g_863_870_fsk_50kbps
#define ANTENNA_SW_SELECT_DEF_CONF      ANTENNA_SW_SELECT_SUBGHZ
#define CC1200_CONF_USE_GPIO2           0
#define CC1200_CONF_USE_RX_WATCHDOG     0
#define CC1200_CONF_802154G             0
#define CC1200_CONF_802154G_CRC16       0
#define CC1200_CONF_802154G_WHITENING   0

#endif /* PROJECT_CONF_H_ */
```
Then, compile for the firefly platform `make TARGET=zoul BOARD=firefly` and run renode in the rpl-udp example directory. Once renode is running, start the `cc1200-test.resc` script with `s @cc1200-test.resc`. You'll quickly notice that the simulation is broken and no messages are exchanged between the simulated nodes (you can check that it does work by commenting out all defines in `project-conf.h`, which will cause the `cc2538_rf_driver` to be used instead, which does work as intended). I suspect it has something to do with the following terminal output:

[image]

It seems that [the firefly platform file](https://github.com/renode/renode/blob/master/platforms/boards/zolertia-firefly.repl) (presumably `zolertia-firefly.repl`) is incomplete or otherwise incorrect. This `.repl` file has the following contents and is supplied in the renode-portable release:

https://github.com/renode/renode/blob/47bc2e11e9e735a46fab9a1390fb06267481f21b/platforms/boards/zolertia-firefly.repl#L1-L9

JakubJatczak, 2021-04-21:
Hi,
Thanks for the report. I reproduced your setup, and can confirm that there are no communication between nodes. I'm currently investigating this issue, so far I can tell that the `Unsupported GPIO mode` logs are unrelated.

JakubJatczak, 2021-04-30:
Sorry to keep you waiting so long. 
There is a mistake in your `.resc` file - all nodes are connected using the `radio` peripheral, which is a CC2538 radio, so try using this line to connect: `connector Connect ssi0.cc1200 wireless`. After that the packets are exchanged as expected ( you can see that in logs after issuing `logLevel -1 wireless` ). 
But even after fixing this connection I was not able to make it work - the frames are received and put in FIFO but they are never read by the software. Seems that while the CC1200 is set to FIFO mode and the software awaits the `cc1200_rx_interrupt`,there is no GPIO configured as `RXFIFO_THR` so the interrupt never comes. While investigating this I found a small bug in the way we handled frames, fix is currently in our internal review and should be merged soon, but it doesn't fix your issue.
Having said that I certainly lack experience with this OS, so if I'm missing something let me know. I'll be glad to make it work

JakubJatczak, 2021-05-14:
Mentioned fixes to `CC1200` are now merged with https://github.com/renode/renode/commit/6fae9e19368148c08429e9f06490ef974f5977d1

mgielda, 2022-02-18:
Closing this since there has been no input from the issue creator for a long time, and the issue seems to be fixed. Reopen if needed.

closed 2022-02-18 by mgielda