renode/renode #191: How can one implement "run until" function without GDB?

resetnow, 2021-03-17:
Hi, first of all thanks for making this project, it's really quite unique and well written.

I'm trying to implement a `repl` script to run an ELF until it reaches `exit`, as in, C runtime function `exit`. Here's what I was able to come up with:

```
$exit = `sysbus GetSymbolAddress "exit"`
$exitScript = """
cpu.Pause()

monitor_class = Antmicro.Renode.UserInterface.Monitor
object_creator = Antmicro.Renode.Core.ObjectCreator.Instance
monitor = object_creator.GetSurrogate(monitor_class)
monitor.HandleCommand('log "Program called exit()"', monitor.Interaction)
monitor.HandleCommand('execute quit', monitor.Interaction)
"""
cpu AddHook $exit $exitScript
```

The problem is, this only works when running renode with GUI. If I use `--disable-xwt`, the python hook handler will still run, but the simulator won't exit. I assume it's because the notion of "quit" is different depending on whether or not there's a terminal emulator. Is there a way to trigger exit from a hook?

resetnow, 2021-04-07:
For anyone stumbling upon this same problem, eventually I solved it by creating a custom command that sends SIGINT to the process. That command is then triggered from the python hook. If you have possibility to patch renode and maintain that patch, see the referenced commit.

mateusz-holenko, 2021-04-19:
FTR, another way of handling this situation is at: https://github.com/antmicro/renode-riscv-torture-example/blob/main/torture.resc. 
It uses a dictionary in the emulation to keep track of the current status and additional python loop to observe it and quit Renode in a proper moment.