renode/renode #197: RISC-V: Memory access is translated, but can't set breakpoints

xobs, 2021-04-09:
Memory access is now translated, meaning the MMU is walked whenever it is enabled. However, this does not appear to be the case for breakpoints.

For example, I have a program that is crashing. I would like to trace this program. When it crashes, it ends up in the kernel and the whole system halts.

Replicating the crash should be as simple as:

1. Figuring out where it crashed
2. Adding a breakpoint
3. Resetting the system

However, I am unable to attach a breakpoint anywhere at all. For example:

``
(gdb) mon machine Reset
(gdb) c
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0xffd0230e in ?? ()
(gdb) mon cpu PRIV 0
(gdb) b _start
Breakpoint 5 at 0x21c76: file /mnt/d/Code/Xous/Core/xous-rs/src/lib.rs, line 169.
(gdb) mon machine Reset
(gdb) c
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0xffd0230e in ?? ()
(gdb)
```

Any address that starts with `0xff______` is inside the kernel. The entrypoint is the usual `_start` function. This program does start, and I can see the output being displayed on the serial console, however the breakpoint is never triggered.

xobs, 2021-04-09:
Update: I can add a breakpoint to both the untranslated address and the translated address, and that causes it to work.

mateusz-holenko, 2021-05-27:
The problem turned out to be related to proper clearing of translation cache.
It should be fixed in: https://github.com/renode/renode/commit/4caf86ac0433d095a6f098fba7e257b8302890a7. 

Could you verify it works for you now, @xobs ?

xobs, 2023-12-28:
Yes, I can confirm that breakpoints work now.

Watchpoints don't seem to work with translated memory. Should they?