renode/renode #199: Breaking changes in the refactored PLIC model

mateusz-holenko, 2021-04-14:
We have recently refactored the PLIC (`PlatformLevelInterruptController`) model in order to make it more robust and useful in different scenarios.

The main difference is replacing the notion of targets (which we understood as a hart with a fixed number of supported privilege levels) with contexts (that might be mapped to any hart/privilege level).

This change results in the need of modifying the platform definition file (`repl`) in order to:
* change the `numberOfTargets` property to `numberOfContexts`
 note: that this is not just a matter of renaming the argument but also updating the value;
* reconnect IRQs, as the identifiers of contexts might not match identifiers used for targets.

We applied the above-mentioned transformations to all platforms currently available in Renode in https://github.com/renode/renode/commit/0d9f10becb98d1135eec1fc979b04ddf566a2585.

Below we present an example of how to perform the transformation for non-public PLIC-based platforms manually.

Assuming that the platform has two RISC-V cores, both supporting machine and supervisor privilege levels, you need to:

* set number of contexts to 4 (2 cores, each supporting 2 privilege levels);
* map PLIC contexts `[0, 1, 2, 3]` to the proper CPU inputs in the repl file
(the mapping depends on the platform, but most probably pins `[0, 1]` will be mapped to CPU0 and `[2, 3]` to CPU1).

mateusz-holenko, 2021-04-14:
CC @xobs @AntonKrug @hehgoog

ImmortalBuka, 2021-05-05:
Where it possible to find memory map for that PLIC?

mateusz-holenko, 2021-05-05:
@ImmortalBuka The layout of registers is defined as an enum in: https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Cores/RiscV/PlatformLevelInterruptController.cs#L76-L102.

closed 2021-12-29 by PiotrZierhoffer