renode/renode #203: Can't install a program compiled with mcuxpresso on the emulated mimxrt1064 evk

DridriLaBastos, 2021-04-23:
I am trying to emulate the nxp mimxrt1064 with renode `1.12` on ` Windows 10 20H2 version 19042.928`. I installed the machine as indicated in the tutorial by loading the script `@platforms/cpus/imxrt1064.repl`, but when comes to the part to install a binary I have some Warnings and I cannot run the simulation : 

[image]

[image]

The installation process runs well with mcuxpresso

[image]

[image]

PiotrZierhoffer, 2021-04-27:
This is a result of Renode not supporting the SPI Flash at the moment.

Here's the code I suggest you use as a loading script that can "hot patch" our platform file with some stub memory region.

This is verified to work with Zephyr. The ELF created by Zephyr build system does not provide the vector table at the lowest load address (which is the case in most situations), so I additionally provide the `VectorTableOffset` value, based on the relevant symbol location.

```
using sysbus
mach create
machine LoadPlatformDescription @platforms/cpus/imxrt1064.repl

# Below I add memory regions where they are missing.
# Please note that it's not a "proper" implementation.

set additional_memory
"""
flash: Memory.MappedMemory @ sysbus 0x70000000
    size: 0x80000

semc: Memory.MappedMemory @ sysbus 0x80000000
    size: 0x80000
"""

machine LoadPlatformDescriptionFromString $additional_memory

$bin?=@zephyr.elf
sysbus LoadELF $bin

# Set the address of the vector_table symbol in your ELF
cpu VectorTableOffset `sysbus GetSymbolAddress "_vector_table"`

showAnalyzer lpuart1
```

DridriLaBastos, 2021-04-28:
Thank you for your help, I used your script and it worked well! I was able to install the ef file and run the emulator, however, I didn't see the `hello world message`. Instead, I have a blank screen. I have the same result when I tested on macOS so maybe I am missing a setting?

[image]

PiotrZierhoffer, 2021-05-11:
I would need some logs (especially with `cpu LogFunctionNames true` in your script) to find out what's going on.

Maybe the vector table offset is different (pointed to by a different symbol)?

You can mail me the binary so that I take a look.