renode/renode #204: Support deriving addresses from SVD

xobs, 2021-04-25:
Renode Platform files support a directive called `ApplySVD`. This enables using an SVD file to report unhandled accesses in order to give them more context.

It would be nice to allow the SVD file to be used in order to fill in missing information, and to allow for under-specifying platform addresses. For example:

```xml
<device>
    <peripherals>
        <peripheral>
            <name>UART</name>
            <baseAddress>0xE0001800</baseAddress>
            <interrupt>
                <name>uart</name>
                <value>0</value>
            </interrupt>
        </peripheral>
    </peripherals>
</device>
```

This could then be imported into the platform file by specifying a peripheral named `uart` with the given driver. The name would need to match, though capitalisation may be different (?).

```
uart: UART.LiteX_UART

sysbus:
    init:
        ApplySVD @soc.svd
```

Renode would then look up the `UART` peripheral to determine what the interrupt and base address should be.

PiotrZierhoffer, 2021-04-25:
The base address is relatively straightforward, but I'm not sure about the interrupts. There are 3 values that have to be encoded:

- source interrupt, number or field name (can be skipped if only one is possible in a peripheral)
- interrupt controller to attach to
- destination number

As far as I know the controller is not something you usually specify (not according to [Keil docs](https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_peripherals.html#elem_interrupt), at least).

xobs, 2021-04-25:
Admittedly, the interrupt feature was more aspirational. The base address is the more interesting and useful part of this request.

PiotrZierhoffer, 2021-04-25:
I think it should be possible to allow the following declaration

```
uart: UART.LiteX_UART @ sysbus
```

and have Renode warn if the address is not filled up before starting the emulation. It would also need to be displayed in a sensible way in the peripherals tree.

xobs, 2021-07-16:
I've created `svd2repl` with the meat of it at https://github.com/betrusted-io/xous-core/blob/main/svd2repl/src/generate.rs

I've added this to our build pipeline, so it generates the repl file directly (e.g. https://github.com/betrusted-io/xous-core/blob/main/emulation/soc/betrusted-soc.repl)

Of particular note is that it generates IRQs. Furthermore, if there is a peripheral with a memory region that matches, it skips creating the memory region and instead passes the information to the peripheral. For example the framebuffer, or our crypto accelerator engine (https://github.com/betrusted-io/xous-core/blob/main/emulation/soc/betrusted-soc.repl#L79-L82):

```
memlcd: Video.BetrustedLCD @ sysbus 0xf000c000
    memAddr: 0xb0000000
    memSize: 0x5c20

engine: Miscellaneous.Engine @ sysbus 0xf001f000
    memAddr: 0xe0020000
    memSize: 0x20000
    -> cpu @ 1016
```