renode/renode #205: Provide Example of Connecting SPI Device

xobs, 2021-04-26:
Renode appears to have the ability to connect two SPI devices together, however it's not clear how to accomplish this or if it's even possible to do under certain circumstances.

Renode should provide an example of how to do this, or document how to connect various peripherals together.

xobs, 2021-04-26:
As an example, I have two Machines (named `EC` and `SoC`):

```
path add $ORIGIN
i @peripherals/com_ec.cs # public class ComEc : ISPIPeripheral, IDoubleWordPeripheral, IKnownSize
i @peripherals/com_soc.cs # public class ComSoC : NullRegistrationPointPeripheralContainer<ISPIPeripheral>, IDoubleWordPeripheral, IKnownSize

mach create "SoC"
LoadPlatformDescriptionFromString 'com: SPI.ComSoC @ sysbus 0xF000C000'
mach clear

mach create "EC"
LoadPlatformDescriptionFromString 'com: SPI.ComEc @ sysbus 0xE0005800'
mach clear
```

`EC` is a SPI peripheral, `SoC` has a registration container that accepts a SPI peripheral. How do I connect one to the other? Using `connector` doesn't appear to work:

```
(monitor) connector Connect SoC.com EC.com
The following methods are available:
 - Void Connect (IEmulationElement connectee, IConnectable connector)
Usage:
 connector MethodName param1 param2 ...
There was an error executing command 'connector Connect SoC.com EC.com'
Parameters did not match the signature
(monitor) 
```

PiotrZierhoffer, 2021-05-18:
To have the `connector` working we need something to connect to.

I would follow the pattern of either UARTHub.cs or GPIOConnector.cs. 

UARTHub is symmetric and quite simple.
The GPIOConnector is a little bit over-complicated, as it allows for two-way communication, but it's asymmetric.

To have an SPIConnector we'd need an asymmetric approach, but only one way. So it would implement `IConnectable<IUART>` and `IConnectable<NullRegistrationPointPeripheralContainer<ISPIPeripheral>>` (this one is not perfect, but should work).

The internal working will be pretty simple, I guess. We need to remember to use the following pattern to handle inter-machine events: `item.GetMachine().HandleTimeDomainEvent` (like in uarthub).

This makes the communication asynchronous - it might not be perfect for all scenarios.

xobs, 2021-06-25:
Updated the issue title to reflect the fact that there is no SPIConnector yet.

xobs, 2021-08-30:
As of the latest commit of Xous (https://github.com/betrusted-io/xous-core/commit/a3d69684c8cc1f7b6c702e590a2d57f20090d559) it now is at a point where it is waiting to be able to communicate across a SPI bus:

```
12:38:07.9127 [WARNING] SoC/sysbus: Write of value 0x2202 to an unimplemented register COM:TX (0xF000D000) generated from SVD.
12:38:07.9137 [WARNING] SoC/sysbus: Read from an unimplemented register COM:STATUS (0xF000D00C), returning a value from SVD: 0x0.
12:38:07.9137 [WARNING] SoC/sysbus: Read from an unimplemented register COM:RX (0xF000D004), returning a value from SVD: 0x0.
12:38:07.9137 [WARNING] SoC/sysbus: Read from an unimplemented register COM:STATUS (0xF000D00C), returning a value from SVD: 0x0.
12:38:07.9146 [WARNING] SoC/sysbus: Write of value 0xF0F0 to an unimplemented register COM:TX (0xF000D000) generated from SVD.
12:38:07.9146 [WARNING] SoC/sysbus: Read from an unimplemented register COM:STATUS (0xF000D00C), returning a value from SVD: 0x0.
12:38:07.9146 [WARNING] SoC/sysbus: Read from an unimplemented register COM:RX (0xF000D004), returning a value from SVD: 0x0.
12:38:07.9146 [WARNING] SoC/sysbus: Read from an unimplemented register COM:STATUS (0xF000D00C), returning a value from SVD: 0x0.
12:38:07.9146 [WARNING] SoC/sysbus: Write of value 0xF0F0 to an unimplemented register COM:TX (0xF000D000) generated from SVD.
12:38:07.9146 [WARNING] SoC/sysbus: Read from an unimplemented register COM:STATUS (0xF000D00C), returning a value from SVD: 0x0.
12:38:07.9157 [WARNING] SoC/sysbus: Read from an unimplemented register COM:RX (0xF000D004), returning a value from SVD: 0x0.
12:38:07.9176 [WARNING] SoC/sysbus: Read from an unimplemented register COM:STATUS (0xF000D00C), returning a value from SVD: 0x0.
12:38:07.9176 [WARNING] SoC/sysbus: Write of value 0xF0F0 to an unimplemented register COM:TX (0xF000D000) generated from SVD.
12:38:07.9176 [WARNING] SoC/sysbus: Read from an unimplemented register COM:STATUS (0xF000D00C), returning a value from SVD: 0x0.
12:38:07.9176 [WARNING] SoC/sysbus: Read from an unimplemented register COM:RX (0xF000D004), returning a value from SVD: 0x0.
```
You can get the image by checking out Xous and building it:

1. `git clone https://github.com/betrusted-io/xous-core.git`
2. `cargo xtask renode-image`
3. `renode emulation/xous-release.resc`

The "SOC" (master) side of the register set is defined at https://ci.betrusted.io/betrusted-soc/doc/com.html and the "EC" (slave) side is defined at https://ci.betrusted.io/betrusted-ec/doc/com.html

mgielda, 2022-02-18:
I believe this works now? Can we close the issue?

xobs, 2022-02-18:
It does work. Thank you very much for the help.

closed 2022-02-18 by xobs

Ristovski, 2026-09-20:
@PiotrZierhoffer Any updates on this? Is there any reason why hasn't been merged in yet?