renode/renode #206: If /opt is a separate volume, deb install causes invalid cross device link

wd5gnr, 2021-04-26:
dpkg: error processing archive renode_1.12.0_amd64.deb (--install):
 error creating hard link './opt/renode/tests/test.sh': Invalid cross-device link

Should use a symlink for this case.

wd5gnr, 2021-04-26:
Workaround (if anyone needs it):
sudo mkdir /usr/tmp
sudo ln -s /usr/tmp /opt/renode
sudo dpkg -i ....
sudo mv /opt/renode /opt/renode0
sudo mv /usr/tmp/* /opt/renode
sudo rm /opt/renode0

mateusz-holenko, 2021-05-05:
@wd5gnr thanks for the report!

We are now discussing options to get rid of symbolic links in packages.
I'll post an update once we have something working.

mateusz-holenko, 2021-05-05:
Now I see I've been mistaken - I thought that the packaging tool we use - `fpm` - generates symbolic links in `/usr/bin`, but that's not true. There are actual scripts there. One of them, however, (`renode-test`) is in fact a hardlink to `/opt/renode/tests/test.sh`.

I don't think we need this hardlink at all, but I need to double-check this with other packages.

mateusz-holenko, 2021-05-26:
With https://github.com/renode/renode/commit/abbd0d31899f1db23ae801f54c9f84ae2edd8930 you should see no more hardlinks in the packagen Linux.

Could you check if it works for you now, @wd5gnr?

PiotrZierhoffer, 2021-08-17:
As we have no links anymore, I'll close this one. Feel free to reopen if needed

closed 2021-08-17 by PiotrZierhoffer