renode/renode #207: Renode requires coreutils for installation

tlechman49, 2021-04-26:
Hi, I received this error on attempting to run renode on Macos Big Sur 11.2.3 and received the following error.

renode file:///Applications/Renode.app/Contents/MacOS/macos_run.command: line 6: realpath: command not foundCannot open assembly './bin/Renode.exe': No such file or directory.Saving session......copying shared history......saving history...truncating history files......completed.

After some googling I found that realpath requires coreutils to work and installed coreutils with homebrew: brew install coreutils

This fixed the problem and I was able to run renode afterwards. I had trouble getting renode to recognize my installation of mono via homebrew and gave up and installed it via the link provided in the readme. It would be really convenient if renode could be installed via homebrew along with its dependencies.

mateusz-holenko, 2021-04-27:
Thanks for reporting an issue, @tlechman49!

We are currently working on getting rid of the realpath dependency from all our scripts. I'll let you know once it's finished and merged to master.

mateusz-holenko, 2021-05-18:
With 7c4559cb724cd5a7b356fa72895254859578275c you should not longer need `realpath` installed in the system to run Renode.

Could you check if it works for you now?

mateusz-holenko, 2021-08-17:
I'm closing this for now, if you encounter problems again please fill another issue.

closed 2021-08-17 by mateusz-holenko