renode/renode #208: libusbip: error: udev_device_new_from_subsystem_sysname failed usbip: error: open vhci_driver

m-ogore, 2021-04-28:
I am trying to used Renode integration for Arduino and try to connect the Renode according to Renode documentation. I run into errors shown in the attached images. when i run the last command i run into this error on the terminal:
libusbip: error: udev_device_new_from_subsystem_sysname failed
usbip: error: open vhci_driver

can i get some help on how to go about this?
[image]

m-ogore, 2021-04-30:
i used the commands on the terminal as follows with the following outputs from the terminal
[image]
and then followed instructions as per renode documentation and still got this error
[image]

mateusz-holenko, 2021-05-05:
This is a progress - right now you seem to have a communication over USB/IP with the Arduino loader.

I was able to reproduce the problem locally and it looks to be related to timeouts in the `bossac` tools used by the Arduino IDE to upload the binary. For me it fails when waiting for a status of one of the SAM-BA commands. I recompiled the `bossac` with  higher timeouts and it started to work fine - investigating the issue further.

m-ogore, 2021-05-06:
Kindly provide the steps to recompile the bossac with higher timeouts on ubuntu linux platforms

mateusz-holenko, 2021-05-07:
I cloned `bossac` sources from: https://github.com/arduino/BOSSA, the `nrf` branch.

Then I changed the line https://github.com/arduino/BOSSA/blob/nrf/src/Samba.cpp#L660 from:
```
_port->timeout(TIMEOUT_QUICK);
```
to
```
_port->timeout(TIMEOUT_NORMAL);
```
and compiled with `make` (you might need to install some dependencies).

Finally I replaced the binary in `~/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino2` with the newly compiled one (you can find it in the `bin` directory of the cloned `BOSSA` folder after building).