renode/renode #211: sysbus.cpu LogFunctionNames true -- not working correctly

tcal-x, 2021-05-05:
Renode seems to not be getting the symbol information.  Or, it doesn't understand mangled C++ names. 

It seems every instruction prints out a log line.
```
14:46:04.4494 [INFO] cpu: Entering function tflite:: at 0x4003CC3C
14:46:04.4495 [INFO] cpu: Entering function tflite:: at 0x4003CC40
14:46:04.4495 [INFO] cpu: Entering function tflite:: at 0x4003CC44
14:46:04.4496 [INFO] cpu: Entering function tflite:: at 0x4003CC48
14:46:04.4496 [INFO] cpu: Entering function tflite:: at 0x4003CC4C
etc.
```
Also, 
```
sysbus FindSymbolAt `sysbus.cpu PC`
```
just returns "tflite::".

If I run 'nm' on the software.elf file, it does seem to have the TFLITE function names, although C++ mangled.

To reproduce in CFU-Playground, 
* clone and run `scripts/setup`
* cd to proj/proj_template_v
* run `make -j8 renode`
* in the monitor, type `sysbus.cpu LogFunctionNames true`
* in the uart, type '1' three times to navigate the menu to run an inference

Even in basic C code, I'm seeing more printouts than necessary:
```
14:52:07.9404 [INFO] cpu: Entering function readchar at 0x4005C3A0
14:52:07.9404 [INFO] cpu: Entering function readchar at 0x4005C3A4
14:52:07.9404 [INFO] cpu: Entering function readchar at 0x4005C388
14:52:07.9404 [INFO] cpu: Entering function uart_read_nonblock (entry) at 0x4005C5FC
14:52:07.9405 [INFO] cpu: Entering function uart_read_nonblock at 0x4005C600
14:52:07.9405 [INFO] cpu: Entering function uart_read_nonblock at 0x4005C604
14:52:07.9405 [INFO] cpu: Entering function uart_read_nonblock at 0x4005C608
14:52:07.9406 [INFO] cpu: Entering function uart_read_nonblock at 0x4005C60C
14:52:07.9406 [INFO] cpu: Entering function uart_read_nonblock at 0x4005C610
14:52:07.9406 [INFO] cpu: Entering function uart_read_nonblock at 0x4005C614
14:52:07.9406 [INFO] cpu: Entering function readchar at 0x4005C38C
14:52:07.9407 [INFO] cpu: Entering function readchar at 0x4005C3A0
14:52:07.9407 [INFO] cpu: Entering function readchar at 0x4005C3A4
```

mateusz-holenko, 2021-05-12:
Thanks for reporting the issue, @tcal!

I confirm that demangling symbol names for the TFLite-based binary does not work as expected. From what I see the problem is related to the anonymous namespace - we need to handle it better.

> Even in basic C code, I'm seeing more printouts than necessary:

Do you mean that you see the same symbol reported many times in a row?
This is an expected behaviour as in fact we are logging each executed block (being a sequence of successive instructions).
You should see that PC values at the end of messages change. If not it might suggest executing a loop.

BTW, some entries have the `(entry)` fragment which means that this is the first instruction in the function.

mateusz-holenko, 2021-05-18:
https://github.com/renode/renode/commit/8dad446ccbbee0f019a307ac214ead9be6d903bd fixes demangling of symbol names.
Could you check if it works fine for you now, @tcal-x.

As for the repeated "Entering function X" log entries we are planning to add an option to limit those (but this is a separate task).

closed 2021-05-25 by PiotrZierhoffer