renode/renode #212: Peripherals: UART / STM32F7_USART fixed frequency

jjmilburn, 2021-05-06:
The [STM32F7_USART](https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/UART/STM32F7_USART.cs#L268) peripheral is *close* to the USART peripheral used by chips like the STM32L0, but with a few differences.

I've modelled an STM32L0_UART peripheral as part of building an STM32L071 `cpu` model (not sure how to contribute that to the to the infrastructure repository if thats desired), but along the way noticed that the `STM32F7_USART` peripheral calculates `BaudRate` using a fixed Apb frequency.

For basic use cases, the `STM32F7_USART` appears drop-in compatible with the STM32L0, but because of the fixed frequency, the `BaudRate` parameter will report an incorrect result.

Propose making frequency used to calculate BaudRate configurable for this peripheral.

```diff
+++ b/src/Emulator/Peripherals/Peripherals/UART/STM32F7_USART.cs
@@ -17,10 +17,11 @@ namespace Antmicro.Renode.Peripherals.UART
     [AllowedTranslations(AllowedTranslation.WordToDoubleWord)] 
     public sealed class STM32F7_USART : BasicDoubleWordPeripheral, IKnownSize, IUART
     {
-        public STM32F7_USART(Machine machine) : base(machine)
+        public STM32F7_USART(Machine machine, uint frequency = 200000000) : base(machine)
         {
             IRQ = new GPIO();
             receiveQueue = new Queue<byte>();
+            this.frequency = frequency;
             DefineRegisters();
         }
 
@@ -43,7 +44,7 @@ public void WriteChar(byte value)
             }
         }
 
-        public uint BaudRate => ApbClock / baudRateDivisor.Value;
+        public uint BaudRate => frequency / baudRateDivisor.Value;
         
         public Bits StopBits
         {
@@ -265,7 +266,7 @@ private void UpdateInterrupt()
 
         private readonly Queue<byte> receiveQueue;
 
-        private const uint ApbClock = 200000000;
+        private readonly uint frequency;
 
         private enum Register
         {
```

PiotrZierhoffer, 2021-05-11:
This is a reasonable approach, we'll update the model accordingly. Thanks for the report!

jjmilburn, 2021-05-12:
Thanks @PiotrZierhoffer ! LMK if there is anything further I can do, and thanks for a great tool here!

PiotrZierhoffer, 2021-05-18:
We definitely do accept contributions, so please feel free to send a PR!

We're about to merge a fix for this issue soon (maybe today).

mateusz-holenko, 2021-05-25:
In https://github.com/renode/renode/commit/1079cde5c50fc53db6c1e4c960777e1e18975f00 we introduced the proposed changes.

Thanks again for the report!

closed 2021-05-25 by mateusz-holenko

IshDeshpa, 2023-07-18:
Could this change be carried over to the STM32F4? It has a similar issue with a fixed APB clock frequency.