renode/renode #215: Possible Memory Corruption in RISC-V

xobs, 2021-05-19:
I'm seeing memory corruption in a RISC-V core. I'm currently trying to come up with a minimal test case. The good news is that it's fully repeatable. The bad news is that tweaking the target executable image in any way mitigates the problem.

## The Smoking Gun

```
(gdb) disassemble $pc-32,+64
Dump of assembler code from 0xffd04698 to 0xffd046d8:
   0xffd04698 <kernel::arch::riscv::mem::map_page_inner+412>:   mv      a0,s4
   0xffd0469c <kernel::arch::riscv::mem::map_page_inner+416>:   li      a1,0
   0xffd046a0 <kernel::arch::riscv::mem::map_page_inner+420>:   auipc   ra,0xffffc
   0xffd046a4 <kernel::arch::riscv::mem::map_page_inner+424>:   jalr    876(ra) # 0xffd00a0c <compiler_builtins::mem::memset>
   0xffd046a8 <kernel::arch::riscv::mem::map_page_inner+428>:   srli    a0,s7,0xc
   0xffd046ac <kernel::arch::riscv::mem::map_page_inner+432>:   andi    a0,a0,1023
   0xffd046b0 <kernel::arch::riscv::mem::map_page_inner+436>:   slli    a0,a0,0x2
   0xffd046b4 <kernel::arch::riscv::mem::map_page_inner+440>:   add     a0,s4,a0
=> 0xffd046b8 <kernel::arch::riscv::mem::map_page_inner+444>:   lw      a1,0(a0)
   0xffd046bc <kernel::arch::riscv::mem::map_page_inner+448>:   andi    a1,a1,1
   0xffd046c0 <kernel::arch::riscv::mem::map_page_inner+452>:   bnez    a1,0xffd04750 <kernel::arch::riscv::mem::map_page_inner+596>
   0xffd046c4 <kernel::arch::riscv::mem::map_page_inner+456>:   srli    a1,s2,0x2
   0xffd046c8 <kernel::arch::riscv::mem::map_page_inner+460>:   and     a1,a1,s6
   0xffd046cc <kernel::arch::riscv::mem::map_page_inner+464>:   or      a1,a1,s5
   0xffd046d0 <kernel::arch::riscv::mem::map_page_inner+468>:   ori     a1,a1,193
   0xffd046d4 <kernel::arch::riscv::mem::map_page_inner+472>:   sw      a1,0(a0)
End of assembler dump.
(gdb) mon sysbus ReadDoubleWord 0x40f00000
0x103C04C7
(gdb) stepi
kernel::arch::riscv::mem::map_page_inner (mm=<optimized out>, pid=..., phys=4026638336, virt=4291690496, req_flags=..., map_user=<optimized out>) at src\arch\riscv/mem.rs:385
385         if unsafe { l0_pt.add(vpn0).read_volatile() } & 1 != 0 {
(gdb) mon sysbus ReadDoubleWord 0x40f00000
0x00000000
(gdb) disassemble $pc-32,+64
Dump of assembler code from 0xffd0469c to 0xffd046dc:
   0xffd0469c <kernel::arch::riscv::mem::map_page_inner+416>:   li      a1,0
   0xffd046a0 <kernel::arch::riscv::mem::map_page_inner+420>:   auipc   ra,0xffffc
   0xffd046a4 <kernel::arch::riscv::mem::map_page_inner+424>:   jalr    876(ra) # 0xffd00a0c <compiler_builtins::mem::memset>
   0xffd046a8 <kernel::arch::riscv::mem::map_page_inner+428>:   srli    a0,s7,0xc
   0xffd046ac <kernel::arch::riscv::mem::map_page_inner+432>:   andi    a0,a0,1023
   0xffd046b0 <kernel::arch::riscv::mem::map_page_inner+436>:   slli    a0,a0,0x2
   0xffd046b4 <kernel::arch::riscv::mem::map_page_inner+440>:   add     a0,s4,a0
   0xffd046b8 <kernel::arch::riscv::mem::map_page_inner+444>:   lw      a1,0(a0)
=> 0xffd046bc <kernel::arch::riscv::mem::map_page_inner+448>:   andi    a1,a1,1
   0xffd046c0 <kernel::arch::riscv::mem::map_page_inner+452>:   bnez    a1,0xffd04750 <kernel::arch::riscv::mem::map_page_inner+596>
   0xffd046c4 <kernel::arch::riscv::mem::map_page_inner+456>:   srli    a1,s2,0x2
   0xffd046c8 <kernel::arch::riscv::mem::map_page_inner+460>:   and     a1,a1,s6
   0xffd046cc <kernel::arch::riscv::mem::map_page_inner+464>:   or      a1,a1,s5
   0xffd046d0 <kernel::arch::riscv::mem::map_page_inner+468>:   ori     a1,a1,193
   0xffd046d4 <kernel::arch::riscv::mem::map_page_inner+472>:   sw      a1,0(a0)
   0xffd046d8 <kernel::arch::riscv::mem::map_page_inner+476>:   li      a0,26
End of assembler dump.
(gdb)
```

In the above snippet, the program counter is sitting on the instruction `lw a1,0(a0)`. The contents of memory address `0x40f00000` is `0x103C04C7`. If we step one instruction and read the contents of memory, we see it has changed to `0x00000000`.

Furthermore, we can see that we did not take an interrupt. Additionally, adding hooks to break or print on memory writes will not trigger. Adding a watchpoint in GDB does not trigger.

Because this memory is changing due to a `load` instruction, I am inclined to believe that this is an emulation bug.

xobs, 2021-05-19:
I have added a testcase repo at https://github.com/xobs/renode-phantom-write

Note that I really think this is memory corruption within Renode, because if I change main memory so that it is added via `machine LoadPlatformDescriptionFromString` instead of `machine LoadPlatformDescription`, then it works just fine.

I am running on Windows using version 5288bcf2415808349a7224c50b083daecdd31a00

PiotrZierhoffer, 2021-05-19:
Just to be sure: your soc.repl has a smaller memory range defined than your platform from string. Is that not a problem? When I corrected it, everything boots fine for me on my Linux machine

xobs, 2021-05-19:
Thank you for pointing that out. It is a problem with the test case, and is probably why the `LoadPlatformDescriptionFromString` invocation worked whereas the `LoadPlatformDescription` did not work.

It's still highly suspicious that memory is getting cleared when doing a `lw`, which should only be modifying registers. It's also concerning that the write isn't getting caught with `tag` commands, or logging commands.

It's interesting that you see the failure as well on your Linux machine. Is there something special about the address `0x40f00000`?

PiotrZierhoffer, 2021-05-19:
Honestly speaking I didn't investigate this yet. The address is not special at all. I'll get back to you when I have more details

xobs, 2021-09-21:
I believe this solves the issue. Thanks!

closed 2021-09-21 by xobs