renode/renode #220: Unable to load  Arduino Nano 33 BLE board

russel90, 2021-06-03:
Dear All,

Unable to Load Arduino_nano_33_ble board.
Please advise.

(monitor) mach create
(machine-0) machine LoadPlatformDescription @platforms/boards/arduino_nano_33_ble.r
epl
There was an error executing command 'machine LoadPlatformDescription @/opt/renode/platforms/boards/arduino_nano_33_ble.repl'
Error E04: Could not resolve type: 'Miscellaneous.NRF52840_PPI'.
At /opt/renode/platforms/cpus/nrf52840.repl:28:6:
ppi: Miscellaneous.NRF52840_PPI @ sysbus 0x4001F000
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^

PiotrZierhoffer, 2021-06-03:
Hi @russel90 

Please verify your submodules are up to date with `git submodule update --init --recursive` and then rebuild.

russel90, 2021-06-03:
Hi @PiotrZierhoffer 

Thank you for your quick response.

After update and rebuild, the script was successfully loaded.
I got an error as blow:

Would you please kindly advise? 

**Monitor:**
(monitor) mach create
(machine-0) machine LoadPlatformDescription @platforms/boards/arduino_nano_33_ble.repl
(machine-0) emulation CreateUSBIPServer
(machine-0) host.usb CreateArduinoLoader sysbus.cpu 0x10000 0 "arduinoLoader"
(machine-0) arduinoLoader WaitForBinary 120 True

**Console:**
06:17:18.1748 [INFO] Loaded monitor commands from: /opt/renode/scripts/monitor.py
06:17:45.0823 [INFO] System bus created.
06:17:45.8302 [INFO] Reading cache
06:17:46.0573 [INFO] sysbus: Loaded SVD: /tmp/renode-3158/526cd5bc-f11a-40bf-bdf1-a2f213bd2179.tmp. Name: nrf52840. Description: nRF52840 reference description for radio MCU with ARM 32-bit Cortex-M4 Microcontroller .
06:17:57.6679 [WARNING] arduinoLoader: Unsupported descriptor type: 0x06 (3)
06:17:57.7438 [WARNING] arduinoLoader: Handling an unimplemented CDC class request: SetLineEncoding

PiotrZierhoffer, 2021-06-08:
@russel90 this looks like another issue, so I'll just reopen this one. We'll investigate it further.

PiotrZierhoffer, 2021-06-22:
This is in fact duplicated in https://github.com/renode/renode/issues/221

PiotrZierhoffer, 2021-06-22:
Actually, it's a different issue. There are some timing issues that we reported to Arduino. You can use the patch we posted here: https://github.com/arduino/BOSSA/pull/1 to have a more consistent experience.