renode/renode #221: Unable to Run Renode Arduino Nano 33 BLE Emulation integrated with Arduino IDE

russel90, 2021-06-06:
**Hi There,** 
As following [the Renode Documentation](https://renode.readthedocs.io/en/latest/host-integration/arduino.html), I test Node Arduino Nano 33 BLE Emulation integrated with Arduino IDE.  it was run on Ubuntu 20.04 in VMware.
When I started the emulation, the uart display  show no message and the log had got some warning messages as below.

Please kindly advise.

**Monitor Display**
(monitor) mach create
(machine-0) machine LoadPlatformDescription @platforms/boards/arduino_nano_33_ble.repl
(machine-0) emulation CreateUSBIPServer
(machine-0) host.usb CreateArduinoLoader sysbus.cpu 0x10000 0 "arduinoLoader"
(machine-0) arduinoLoader WaitForBinary 120 True
Binary of size 176128 bytes loaded at 0x10000
(machine-0) 
(machine-0) showAnalyzer sysbus.uart0
(machine-0) start
Starting emulation...
(machine-0) 

**machine-0:sysbus.uart0**

**Monitor Log**
1:51:34.5670 [INFO] Loaded monitor commands from: /opt/renode/scripts/monitor.py
11:51:55.6184 [INFO] System bus created.
11:51:56.3559 [INFO] Reading cache
11:51:56.5821 [INFO] sysbus: Loaded SVD: /tmp/renode-4490/2f3af9e6-1a0a-4cf2-ac0f-d703507939c9.tmp. Name: nrf52840. Description: nRF52840 reference description for radio MCU with ARM 32-bit Cortex-M4 Microcontroller .
11:52:06.9057 [WARNING] arduinoLoader: Unsupported descriptor type: 0x06
11:52:06.9069 [WARNING] arduinoLoader: Unsupported descriptor type: 0x06
11:52:06.9079 [WARNING] arduinoLoader: Unsupported descriptor type: 0x06
11:52:06.9583 [WARNING] arduinoLoader: Handling an unimplemented CDC class request: SetLineEncoding
11:52:16.0727 [WARNING] arduinoLoader: Handling an unimplemented CDC class request: SetControlLineState
11:52:16.1192 [WARNING] arduinoLoader: Handling an unimplemented CDC class request: SetLineEncoding
11:52:16.1356 [WARNING] arduinoLoader: Handling an unimplemented CDC class request: SetControlLineState
11:52:16.1841 [WARNING] arduinoLoader: Handling an unimplemented CDC class request: SetControlLineState
11:52:22.3069 [WARNING] arduinoLoader: Handling an unimplemented CDC class request: SetControlLineState
11:52:22.3648 [WARNING] arduinoLoader: Handling an unimplemented CDC class request: SetLineEncoding
11:52:27.6310 [WARNING] arduinoLoader: Handling an unimplemented CDC class request: SetControlLineState
11:54:06.0934 [INFO] cpu: Setting initial values: PC = 0x241ED, SP = 0x20040000.
11:54:06.0956 [INFO] machine-0: Machine started.
11:54:06.1434 [WARNING] sysbus: [cpu: 0x24258] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x0.
11:54:06.1440 [WARNING] sysbus: [cpu: 0x24258] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x0.
11:54:06.1443 [WARNING] sysbus: [cpu: 0x24258] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x0.
11:54:06.1445 [WARNING] sysbus: [cpu: 0x24258] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x0.
11:54:06.1448 [WARNING] sysbus: [cpu: 0x24258] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x0.
11:54:06.1451 [WARNING] sysbus: [cpu: 0x24258] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x0.
11:54:06.1453 [WARNING] sysbus: [cpu: 0x24258] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x0.
11:54:06.1473 [WARNING] sysbus: [cpu: 0x2438E] (tag: 'PSELRESET') ReadDoubleWord from non existing peripheral at 0x10001200, returning 0x0.
11:54:06.1480 [WARNING] sysbus: [cpu: 0x243B0] ReadDoubleWord from non existing peripheral at 0x10001204.
11:54:06.1685 [WARNING] rtc1: Unhandled write to offset 0x104, value 0x0.
11:54:06.1686 [WARNING] rtc1: Unhandled read from offset 0x104.
11:54:06.1686 [WARNING] rtc1: Unhandled write to offset 0x348, value 0x20002.
11:54:06.1717 [WARNING] rtc1: Unhandled write to offset 0x304. Unhandled bits: [1] when writing value 0x2. Tags: OVRFLW (0x1).
11:54:06.1968 [WARNING] timer1: Unhandled read from offset 0x508.
11:54:06.1969 [WARNING] timer1: Unhandled write to offset 0x508, value 0x3.
11:54:06.2059 [WARNING] gpio0: Unhandled write to offset 0xA24, value 0x0.
11:54:06.2061 [WARNING] gpio0: Unhandled write to offset 0xA24, value 0xC.
11:54:06.2064 [WARNING] gpio0: Unhandled read from offset 0x804.
11:54:06.2070 [WARNING] gpio0: Unhandled write to offset 0x80C, value 0x200.
11:54:06.2071 [WARNING] gpio0: Unhandled write to offset 0xA24, value 0x3.

PiotrZierhoffer, 2021-06-22:
Did

PiotrZierhoffer, 2021-06-22:
@russel90 did you remember to switch the serial output in your code?

    sed -i'' '/#define DEBUG_SERIAL_OBJECT/s/(Serial)/(Serial1)/' ~/Arduino/libraries/Arduino_TensorFlowLite/src/tensorflow/lite/micro/arduino/debug_log.cpp

Can you please run the following before the `start` command, and post the output log?

    cpu LogFunctionNames true