renode/renode #222: BeagleV: Is Ethernet working?

peyo-hd, 2021-06-10:
By following https://github.com/antmicro/renode-beagle-v/blob/master/README.md,
Boot-up and Getting console prompt was successful.

Then, I can see eth0 with 'ifconfig -a' command.
Can we establish network to host?

# ifconfig -a
eth0 Link encap:Ethernet HWaddr 66:34:B0:6C:DE:AD
BROADCAST MULTICAST MTU:1500 Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)
Interrupt:18
lo Link encap:Local Loopback
inet addr:127.0.0.1 Mask:255.0.0.0
UP LOOPBACK RUNNING MTU:65536 Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)

# ifconfig eth0 up
[ 3630.429445] stmmaceth 10020000.gmac eth0: validation of rgmii-txid with support 0000000,00000000,00006280 and advertisement 0000000,00000000,00006280 failed: -22
[ 3630.429880] stmmaceth 10020000.gmac eth0: stmmac_open: Cannot attach to PHY (error: -22)
ifconfig: SIOCSIFFLAGS: Invalid argument

Issue is moved from https://github.com/antmicro/renode-beagle-v/issues/3

PiotrZierhoffer, 2021-06-15:
Hi @peyo-hd, thanks for the report.

While we have an ethernet model, there's apparently a problem with the PHY interaction and the interrupt flow. We'll look into it.

For future updates it's best to follow the main Renode repository, as the sample is already a part of it. Having said that, the problem persists there as well.

rminnich, 2023-09-15:
it's been a while, but this bug persists. Any idea what to do?

rminnich, 2023-09-21:
> Hi @rminnich <https://github.com/rminnich>
>
> We do have a fix for phy, but there are still issues in communication -
> we're looking into this

any luck? Any way we can help? We're a bit stuck.

mateusz-holenko, 2024-05-17:
The BeagleV platform ethernet communication should work fine in the latest version of Renode. We ship automatic tests running ping commands between simulated Linux boards: https://github.com/renode/renode/blob/master/tests/platforms/BeagleV-StarLight.robot.

Please also note that all necessary components are now part of Renode upstream, so the original https://github.com/antmicro/renode-beagle-v/ repository will soon contain just a README file.

Hope this resolves problems you encountered.

closed 2024-12-04 by mateusz-holenko