renode/renode #223: Adding USART3 to stm32f746

ananda-emfios, 2021-06-17:
Hi, 

what changes need to make to enable additional USART interface?

USART1 is already exists in renode/platforms/cpus/stm32f746.repl.

```
usart1: UART.STM32F7_USART @ sysbus 0x40011000
    -> nvic@37

```
I need to add USART3. I modified the .repl as follows.

```
usart3: UART.STM32F7_USART @ sysbus 0x40048000
    -> nvic@39
```
With this change, I am getting the following error.
`sysbus: Write of value 0x0 to an unimplemented register USART3:CR1 (0x40004800) generated from SVD. (2)`

Am I missing any other places to modify?
Do I need to build Renode source to include additional USART?

Thanks for any suggestion.

mateusz-holenko, 2021-06-22:
Hi @ananda-emfios ,

there is no need to recompile Renode after modyfing the repl file. Adding new UART device works exactly as you explained.

I see that there is an inconsistency in addresses:
* `0x4004 8000` is where you install usart3, 
* but the warning says about `0x4000 4800`
 
Isn't this the root problem?

ananda-emfios, 2021-06-22:
Yes! you are right about the address. I verified with document and corrected it. Now USART3 working as expected. My bad, I overlooked at it.
```
usart3: UART.STM32F7_USART @ sysbus 0x40004800
    -> nvic@39
```
Thanks a lot!

closed 2021-06-22 by ananda-emfios