renode/renode #224: Mbed OS sleep APIs not working

davidalo, 2021-06-18:
I am trying to run Mbed OS application using Renode (docker version) over Linux:
```
Renode, version 1.12.0.37823 (44d6786a-202104022100)
```

I saw some issues (such as #184  and #75) which talks about Mbed OS support. After trying Discovery board without success, I am currently trying to use NRF52840 target. The problem is that sleep APIs are not working as expected.

The steps I have done are:

1. Compile an mbed example with the following code:
```cpp
#include "mbed.h"

int main()
{
    while (true) {
        printf("Hello from UART\r\n");
        ThisThread::sleep_for(BLINKING_RATE);
    }
}
```
Using:
``` 
mbed compile --profile debug -m NRF52840_DK
```

2. Create a new machine in Renode, flash the ELF and run:
```
$name="nrf52840"
mach create $name

machine LoadPlatformDescription @platforms/cpus/nrf52840.repl

showAnalyzer uart0

$bin = @/home/developer/mbed-os-example-blinky/BUILD/NRF52840_DK/GCC_ARM-DEBUG/mbed-os-example-blinky.elf
sysbus LoadELF $bin

start
```
3. The result is that `printf` trace is shown in analyzer but only once. Renode log shows:
```
....
10:13:18.6001 [WARNING] nvic: Unhandled value written to System Control Register: 0x10.
10:13:18.6001 [WARNING] clock: Unhandled write to offset 0x7C, value 0x1.
10:13:18.6002 [WARNING] nvic: Unhandled value written to System Control Register: 0x10.
10:13:18.6002 [WARNING] clock: Unhandled write to offset 0x7C, value 0x1.
```

In addition to this, I have tried to use Mbed OS wait API by using: `wait_us` instead of `sleep_for`. This wait_us uses a LPTIM to wait. It neither works. Seems that Low Power timer are not implemented in Renode
```
...

09:36:33.7265 [WARNING] sysbus: Write of value 0x1 to an unimplemented register LPTIM1:CR (0x40002410) generated from SVD.
09:36:33.7268 [WARNING] sysbus: Write of value 0x10 to an unimplemented register LPTIM1:ICR (0x40002404) generated from SVD.
09:36:33.7270 [WARNING] sysbus: Write of value 0xFFFF to an unimplemented register LPTIM1:ARR (0x40002418) generated from SVD.
09:36:33.7349 [WARNING] sysbus: Read from an unimplemented register LPTIM1:CR (0x40002410), returning a value from SVD: 0x0.
09:36:33.7353 [WARNING] sysbus: Write of value 0x4 to an unimplemented register LPTIM1:CR (0x40002410) generated from SVD.
09:36:33.7355 [WARNING] sysbus: Write of value 0x8 to an unimplemented register LPTIM1:ICR (0x40002404) generated from SVD.
09:36:33.7356 [WARNING] sysbus: Write of value 0x0 to an unimplemented register LPTIM1:CMP (0x40002414) generated from SVD.
09:36:33.7377 [WARNING] sysbus: Write of value 0x8 to an unimplemented register LPTIM1:ICR (0x40002404) generated from SVD.
09:36:33.7510 [WARNING] sysbus: Read from an unimplemented register LPTIM1:CNT (0x4000241C), returning a value from SVD: 0x0. (8)
09:36:33.7511 [WARNING] sysbus: Write of value 0x1 to an unimplemented register LPTIM1:ICR (0x40002404) generated from SVD.
09:36:33.7513 [WARNING] sysbus: Write of value 0x7000 to an unimplemented register LPTIM1:CMP (0x40002414) generated from SVD.
09:36:33.7546 [WARNING] sysbus: Read from an unimplemented register LPTIM1:CNT (0x4000241C), returning a value from SVD: 0x0. (10)
```

It would be nice to complete the support for Mbed OS for any board. I have extense experience in Mbed OS but I am new in Renode, if you can help me with it I can try to complete the support for Mbed OS and add some examples .

Thanks in advance

davidalo, 2021-06-21:
I have created a new public repository in order to track the development of the Mbed OS integration with Renode, it includes instructions about how to run this issue from scratch. See [renode-mbed-example](https://gitlab.com/davidalo/renode-mbedos-example)

davidalo, 2021-06-24:
I found the Offical Mbed OS Renode example in the [testing renode web](https://testing.renode.io/), I will try to reproduce it and compare with my problems. I will give you feedback asap.

PiotrZierhoffer, 2021-06-24:
Hi @davidalo , thanks for the report.

I can see some unimplemented registers in the rtc and timer, so it might be the cause. I believe that unimplemented RTC overflow might be the root cause here.

Can you please add a log with the following commands in the script you run?

```
cpu LogFunctionNames true
sysbus LogPeripheralAccess rtc1
sysbus LogPeripheralAccess timer1
sysbus LogPeripheralAccess ppi
```
Please note that the last command will work on a fairly recent Renode clone, as it's a new peripheral we just added. If you don't have it then it's not a problem.

davidalo, 2021-06-24:
Hi @PiotrZierhoffer, thanks you for your support.

I have run over NRF52840 with the following commands (as you said, `ppi` was not available):
```
$name="nrf52840"

mach create $name

machine LoadPlatformDescription @platforms/cpus/nrf52840.repl

using sysbus
showAnalyzer sysbus.uart0
cpu LogFunctionNames true
sysbus LogPeripheralAccess rtc1
sysbus LogPeripheralAccess timer1
sysbus LogPeripheralAccess ppi

# wait application
$bin = @/home/developer/mbed/BUILD/NRF52840_DK/GCC_ARM-DEBUG/wait.elf

sysbus LoadELF $bin
start
```

I attach you the log I got from the docker output (I captured some seconds):
 
[log.txt](https://github.com/renode/renode/files/6708727/log.txt)

davidalo, 2021-06-25:
I have tested the ["official" example repo](https://github.com/andresR8/mbed-os-renode-example) and the behaviour is the same, it does not work with these APIs.

davidalo, 2021-07-20:
Any update on this @PiotrZierhoffer ?

PiotrZierhoffer, 2021-07-20:
Actually yes, we have a person looking into this. We will try to update you as soon as we have details

mateusz-holenko, 2021-08-17:
Hi @davidalo, 

We debugged the problem and it seems Renode currently lacks support for the `WFE` instruction. As a result some fragments of code loop actively instead of sleeping which results in a low performance of the simulation so from the user perspective it feels like sleep never ends.

We are implementing missing pieces.

davidalo, 2021-08-18:
Hi @mateusz-holenko,

Thanks for you response. I wil wait until news.

davidalo, 2021-10-19:
Hi @mateusz-holenko 

In my first tests, `sleep_for` is working as expected. The execution is a bit slower than the real time but the internal timing seems to work as expected. The following example prints a message every 5 seconds, it does it every 9 seconds in host machine (more or less) but internal timing seems correct.

```cpp
#include "mbed.h"

#define BLINKING_RATE_MS     5000

int main()
{
    while (true) {
        printf("%llu - Hello from UART\r\n", time(NULL));
        ThisThread::sleep_for(BLINKING_RATE_MS);
    }
}
```

Output log is:
```
0 - Hello from UART
5 - Hello from UART
10 - Hello from UART
```

In addition to this, if I use `wait_us` (with LPTimer) it does not work. Log is available 
[here](https://github.com/renode/renode/files/7371723/renode-mbed-os-example-log-wait.txt)

Renode version: Current master version in docker executor.