renode/renode #225: Support truncating output with `CreateServerSocketTerminal`

xobs, 2021-06-25:
When creating a socket with something such as `CreateServerSocketTerminal`, the output is buffered. That is, when a client connects, the entire history of the UART (up to the size of the UART buffer) is returned to the client. Ordinarily this is what you want.

However, when starting something such as a GDB server, this is undesirable.

For example, say a serial pipe has a command line that can be switched to speak GDB's `RSP`. When gdb connects to the socket over the network, it will get the entire history of the UART before it gets to the part containing RSP commands.

It would be nice to have a way to ensure the socket is flushed such that the client receives no data when it connects.

PiotrZierhoffer, 2021-06-29:
Hi @xobs 
At the moment it does not seem to be very easy, because of some architectural details.

We will look more at this, but it might take a while.

electrodonkey, 2023-06-15:
Hello, is there any update on this?
In my usecase (testing custom serial protocol over UART) the receiving application connected via VSP (HW VSP3) is confused when receiving the entire buffer dump.
Would be nice to have the option to disable buffering as @xobs mentioned.