renode/renode #226: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast] for v0.12.0 on Windows 10 x86 using Cygwin

andreok, 2021-06-28:
Hello,

I am getting many errors like the following when trying to build v0.12.0 on Windows 10 x86, using Cygwin:
`...renode\src\Infrastructure\src\Emulator\Cores\translate.cproj(126,5): error : tlib/tcg
/tcg-op.h:391:24: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]`
I tried to add `CFLAGS=-Wno-error=pointer-to-int-cast` to the PARAMS inside build.sh, but it didn't help.

Regards,
Andre

PiotrZierhoffer, 2021-06-29:
Hi @andreok 

Do I understand correctly that you are trying to build Renode on a 32-bit machine?

This is not currently supported, you will need a 64-bit host to use Renode.

PiotrZierhoffer, 2021-08-17:
Closing this one, but feel free to reopen if you're on 64-bit machine.

closed 2021-08-17 by PiotrZierhoffer