renode/renode #229: CAN bus support example (Zephyr)

bmeisels, 2021-07-10:
I have been working on getting a Zephyr CAN network app working on renode.
I am using the https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/sockets/can sample which I have adapted to use within Platformio (https://github.com/bmeisels/STM32_Zephyr_Renode).

In order to use this sample I modified the stm32f4 discovery configuration to include CAN2 (which is the default for this board in zephyr).

Next I had the following crash which is fixed by https://github.com/renode/renode-infrastructure/pull/36

> Object reference not set to an instance of an object
  at Antmicro.Renode.Peripherals.CAN.STMCAN+CANMessage..ctor (Antmicro.Renode.Core.CAN.CANMessageFrame message) [0x0001e] in <80dc79c9f02e4bc3b127d435fa694291>:0 
  at Antmicro.Renode.Peripherals.CAN.STMCAN.OnFrameReceived (Antmicro.Renode.Core.CAN.CANMessageFrame message) [0x00095] in <80dc79c9f02e4bc3b127d435fa694291>:0 
  at Antmicro.Renode.Core.Machine+<>c__DisplayClass45_0`1[T].<HandleTimeDomainEvent>b__1 () [0x00000] in <49c9daf93fdb4767852aafd4882a5b41>:0 
  at Antmicro.Renode.Core.Machine+<>c__DisplayClass113_0.<ReportForeignEventInner>b__0 (Antmicro.Renode.Time.TimeStamp ts) [0x0001e] in <49c9daf93fdb4767852aafd4882a5b41>:0 
  at Antmicro.Renode.Time.TimeSourceBase.ExecuteInNearestSyncedState (System.Action`1[T] what, System.Boolean executeImmediately) [0x0002d] in <49c9daf93fdb4767852aafd4882a5b41>:0 
  at Antmicro.Renode.Core.Machine.ReportForeignEventInner (System.Action`2[T1,T2] recordMethod, System.Action handlerMethod, System.Boolean timeDomainInternalEvent) [0x00021] in <49c9daf93fdb4767852aafd4882a5b41>:0 
  at Antmicro.Renode.Core.Machine.HandleTimeDomainEvent[T] (System.Action`1[T] handler, T handlerArgument, System.Boolean timeDomainInternalEvent) [0x0001b] in <49c9daf93fdb4767852aafd4882a5b41>:0 
  at Antmicro.Renode.Core.Machine+<>c__DisplayClass43_0`1[T].<HandleTimeDomainEvent>b__0 (Antmicro.Renode.Time.TimeStamp ts) [0x00029] in <49c9daf93fdb4767852aafd4882a5b41>:0 
  at Antmicro.Renode.Time.TimeSourceBase.ExecuteSyncPhase () [0x0008d] in <49c9daf93fdb4767852aafd4882a5b41>:0 
  at Antmicro.Renode.Time.TimeSourceBase.InnerExecute (Antmicro.Renode.Time.TimeInterval& virtualTimeElapsed, System.Nullable`1[T] timeLimit) [0x0025f] in <49c9daf93fdb4767852aafd4882a5b41>:0 
  at Antmicro.Renode.Time.SlaveTimeSource.DispatchInner () [0x00039] in <49c9daf93fdb4767852aafd4882a5b41>:0 
  at Antmicro.Renode.Time.SlaveTimeSource.Dispatch () [0x00043] in <49c9daf93fdb4767852aafd4882a5b41>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) [0x00017] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x0008d] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x00031] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Threading.ThreadHelper.ThreadStart () [0x0000b] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 

Compiling Renode from source solved this issue.

After these changes the output on both boards is the following (The send loop continues forever).

> *** Booting Zephyr OS build zephyr-v20500  ***
[00:00:02.000,000] <dbg> net_sock_can.can_register_filters: (main): Registering 1 filters
[00:00:02.000,000] <dbg> net_sock_can.can_register_receiver: (main): Max 1 receivers
[00:00:02.000,000] <dbg> can_driver.can_stm32_set_filter: Setting filter ID: 0x1, mask: 0x7ff
[00:00:02.000,000] <dbg> can_driver.can_stm32_set_filter: Filter type: standard ID 
without mask (1)
[00:00:02.000,000] <dbg> can_driver.can_stm32_set_filter: Filter set! Filter number
: 0 (index 0)
[00:00:02.000,000] <dbg> net_socket_can_sample.setup_socket: Started socket CAN TX 
thread
[00:00:02.000,000] <inf> net_socket_can_sample: 1st RX fd 0
[00:00:02.000,000] <dbg> net_socket_can_sample.rx: [0] Waiting CAN data...
[00:00:03.000,000] <dbg> net_socket_can_sample.tx: Sending CAN data...
[00:00:03.000,000] <dbg> can_driver.can_stm32_send: Sending 8 bytes on CAN_2. Id: 0
x1, ID type: standard, Remote Frame: no
[00:00:03.000,000] <dbg> can_driver.can_stm32_send: Using mailbox 0
[00:00:04.000,000] <dbg> can_driver.can_stm32_send: Sending 8 bytes on CAN_2. Id: 0
x1, ID type: standard, Remote Frame: no

It seems that messages are never revived.
I don't have hardware on hand to test so I am not sure if this is a configuration issue or an issue with Renode (or something I overlooked in the code).

Would appreciate if anyone could point me in the right direction to solve this.

mateusz-holenko, 2021-08-31:
Hi @bmeisels, thanks for reporting the issue!

With c7a5ccf6e17ab22d5da085116dfdf741db1b8f70 upstream you should be able to run the demo on Zephyr v2.5. We noticed problems on newer Zephyr, but it seems to be software bug (not the simulation one).

Could you verify if it works for you now?

bmeisels, 2021-08-31:
Awesome!

Thank you so much for lookin at this.
Can confirm this works with Zephyr v2.5 and as mentioned not on Zephyr v2.6.
I don't have a board to test to see if this is the case for IRL but will try to get one for testing.

MateuszKarlic, 2021-09-27:
Hi @bmeisels we reported the bug with newer Zephyr versions on their issue tracker. You can view its status here: https://github.com/zephyrproject-rtos/zephyr/issues/38698

closed 2021-09-27 by MateuszKarlic