renode/renode #231: GDB commands not working properly

Ahelion, 2021-07-13:
Using SystemWorkbench, configured for STM32 debug,
Renode running with a stm32 elf.
When debuging step over does not work, each function is entered, and chunks of code of 'skipped".
Everything is sluggish.
I debuged it a bit, on the gdb server socket aparenlty all data is send byte by byte, and i think that may be the root cause.
Other Gdb servers( like the one in qemu), send packets consisting of several bytes.
Did anybody else notice it?
This is the only command i use:
machine StartGdbServer 1234

PiotrZierhoffer, 2021-08-03:
Hi @Ahelion, thanks for the report.

Can you please provide some GDB logs? You can enable them with `(gdb) set debug remote 1`.

It might be a specific SystemWorkbench behavior - stepping over functions is handled by GDB itself, not by the protocol we talk to it.

PiotrZierhoffer, 2021-08-17:
@Ahelion we recently pushed some changes to GDB handling - can you please verify if they work for you?

Ahelion, 2021-11-11:
> @Ahelion we recently pushed some changes to GDB handling - can you please verify if they work for you?

Hello, I tried the latest renode today, but the debug experience is the same. Functions are entered each time, and everything is very slow.

Ablu, 2021-11-11:
In our case we do not run into this issue if we test it on Windows with the arch-specific gdb. So maybe this either is a Linux or gdb-multiarch issue...

monikaveeravalli, 2024-05-29:
hi i am not able to even connect to gdb the gdb window or the gdb setup is not not appearing after i  give 
machine StartGdbServer 3333 
when i am giving an elf file it says 
arm-none-eabi-gdb /path/to/application.elf 
arm-none-eabi-gdb is not found