renode/renode #232: Adding support of the STM32L4Axx

Emna-b, 2021-07-16:
I am working on adding a .repl file of the mcu STM32L4xx and I need to know wether I need to add a model for peripherals (uart,usart,gpio..) or the existing ones(those of smt32f7) can do the job ?

PiotrZierhoffer, 2021-08-03:
ST reuses a lot of IP, so there's a high chance you can reuse them - I'd recommend comparing the register maps for relevant peripherals for [STM32L4x](https://www.st.com/resource/en/reference_manual/dm00151940-stm32l41xxx42xxx43xxx44xxx45xxx46xxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) and [STM32F7x](https://www.st.com/resource/en/reference_manual/dm00124865-stm32f75xxx-and-stm32f74xxx-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf).

Also, STM32F4 has some differences to F7, and we implement both versions of peripherals - so either F4 or F7 could be usable in your platform.

yorange1, 2021-09-01:
> ST reuses a lot of IP, so there's a high chance you can reuse them - I'd recommend comparing the register maps for relevant peripherals for [...]

It is really difficult to add a new MCU support for renode. I hope that there will be documents or tutorials to guide developers to add more MCUs and peripherals to Renode.

mithro, 2021-09-06:
@yorange1  - Having added new platforms to both QEMU and Renode, Renode is about a bazillion times easier. See also @xobs' experience at https://xobs.io/precursor-and-renode/. 

There is also a large amount of documentation at docs.renode.io including a guide to describing platforms at https://renode.readthedocs.io/en/latest/basic/describing_platforms.html -- Yes, it is actually as easy as writing a single text file.

yorange1, 2021-09-07:
> @yorange1 - Having added new platforms to both QEMU and Renode, Renode is about a bazillion times easier. See also @xobs' experience at https://xobs.io/precursor-and-renode/.
>
> There is also a large [...]

Thank you, Renode is indeed much better than QEMU. I have found that my problem comes from learning MCU and peripherals instead of Renode. It is not easy to write a "single text file" that can work correctly. Thank Renode has supported so many platforms.

xobs, 2021-09-07:
I added a chapter to the `Xous Book` on how to add C# peripherals: https://betrusted.io/xous-book/ch04-04-writing-cs-peripherals.html.

It needs more work to add examples on things like peripherals with memory ranges, but it should be a good starting point.