renode/renode #233: GDB step over becomes step into

hcindyl, 2021-07-28:
Use Renode + gdb for a RISCV32 target. If I use "next" to step over functions, it actually steps into the function, and I would need to explicitly step out of the function with "finish". It seems to be the same behavior as "step"

mateusz-holenko, 2021-08-10:
Hi @hcindyl, thanks for reporting the issue!

We found a problem in a generation of registers list that was reported to GDB. This lead to a wrong interpretation of exchanged 
packets and caused general registers (x1 - x31) to have wrong values according to GDB.
This could be a root cause of problems with stepping over functions and backtracking.

Could you check if eb0f1449086ade58f43ba23c986bf45972d79185 solves the problem for you?

hcindyl, 2021-08-10:
Thanks Mateusz for the fix. Yes it seems to address the GDB problems we encounter, including step over, register value query, and backtrace.

closed 2021-08-11 by mateusz-holenko