renode/renode #234: Failed Building the Renode.sln on Visuan Studio

Emna-b, 2021-07-31:
I was setting up the Renode build environment, and when I tried to build the Renode.sln file using Visual Studio I got multiple errors. (This step is mentioned in the Renode Documentation)
Errors : 
-error with the code : CS0246  
-components not found
Any help would be appreciated! 

[image]

PiotrZierhoffer, 2021-08-03:
When building Renode for the first time on each clone you need to run `./build.sh`. This should

a) download all submodules
b) download additional build dependencies.

If this fails for you, please provide the full output of build.sh

Emna-b, 2021-08-03:
When I ran ./build.sh on Cygwin this are the errors I got : 
./build.sh: line 58: xbuild: command not found

[image]

PiotrZierhoffer, 2021-08-17:
I'm not sure why do you run the first command `cd ./lib/Migrant/tools`? This is just a helper script, it won't work and you don't need it.

Just run `./build.sh` from the top Renode dir.

PiotrZierhoffer, 2021-09-21:
Let's close this for now, please reopen if you have any other comments or questions

closed 2021-09-21 by PiotrZierhoffer