renode/renode #235: Fix WARNINGS when writing to some registers - STM32F0

MacAndKaj, 2021-08-02:
- Platform: Linux
- renode: master@0db5692aeffe3e46985415a48b045b3e2853459f
- [elf_and_renode_files.zip](https://github.com/renode/renode/files/6919819/elf_and_renode_files.zip)

Using renode for STM32F072RBT6 still getting warnings like

08:46:53 [WARNING] sysbus: [cpu: 0x800073A] (tag: 'SYSCFGCOMP') ReadDoubleWord from non existing peripheral at 0x40010014, returning 0x0.
08:46:53 [WARNING] sysbus: [cpu: 0x8000626] (tag: 'SYSCFGCOMP') WriteDoubleWord to non existing peripheral at 0x40010014, value 0x20.
08:46:53 [WARNING] gpioPortA: Unhandled write to offset 0x4. Unhandled bits: [6] when writing value 0x40. Tags: OT6 (0x1).
08:46:53 [WARNING] gpioPortA: Unhandled write to offset 0x4. Unhandled bits: [8] when writing value 0x100. Tags: OT8 (0x1).
08:46:53 [WARNING] gpioPortA: Unhandled write to offset 0x4. Unhandled bits: [9] when writing value 0x200. Tags: OT9 (0x1).
08:46:53 [WARNING] gpioPortC: Unhandled write to offset 0x4. Unhandled bits: [7] when writing value 0x80. Tags: OT7 (0x1).
08:46:53 [WARNING] sysbus: [cpu: 0x8000610] (tag: 'SYSCFGCOMP') ReadDoubleWord from non existing peripheral at 0x40010010, returning 0x0.
08:46:53 [WARNING] sysbus: [cpu: 0x8000610] (tag: 'SYSCFGCOMP') WriteDoubleWord to non existing peripheral at 0x40010010, value 0x2.
08:46:53 [WARNING] sysbus: [cpu: 0x8000692] (tag: 'SYSCFGCOMP') ReadDoubleWord from non existing peripheral at 0x40010010, returning 0x0.
08:46:53 [WARNING] sysbus: [cpu: 0x8000692] (tag: 'SYSCFGCOMP') WriteDoubleWord to non existing peripheral at 0x40010010, value 0x200.
08:46:53 [WARNING] gpioPortB: Unhandled write to offset 0x4. Unhandled bits: [6] when writing value 0x40. Tags: OT6 (0x1).
08:46:53 [WARNING] gpioPortA: Unhandled write to offset 0x20. Unhandled bits: [8] when writing value 0x100. Tags: AFSEL2 (0x1).
08:46:53 [WARNING] gpioPortA: Unhandled write to offset 0x20. Unhandled bits: [12] when writing value 0x1000. Tags: AFSEL3 (0x1).
08:46:53 [WARNING] timer2: Unhandled write to offset 0x4. Unhandled bits: [5] when writing value 0x20. Tags: MMS (0x2).
08:46:53 [WARNING] timer2: Unhandled write to offset 0x1C. Unhandled bits: [5-6] when writing value 0x60. Tags: RESERVED (0x18).
08:46:53 [WARNING] timer2: Unhandled write to offset 0x1C. Unhandled bits: [3] when writing value 0x8. Tags: RESERVED (0x2).
08:46:53 [WARNING] gpioPortB: Unhandled write to offset 0x4. Unhandled bits: [10] when writing value 0x400. Tags: OT10 (0x1).
08:46:53 [WARNING] gpioPortB: Unhandled write to offset 0x24. Unhandled bits: [9] when writing value 0x200. Tags: AFSEL10 (0x2).
08:46:53 [WARNING] sysbus: [cpu: 0x8005B20] (tag: 'SYSCFGCOMP') ReadDoubleWord from non existing peripheral at 0x40010000, returning 0x0.
08:46:53 [WARNING] sysbus: [cpu: 0x8005B20] (tag: 'SYSCFGCOMP') WriteDoubleWord to non existing peripheral at 0x40010000, value 0x40000000.
08:46:53 [WARNING] timer3: Unhandled write to offset 0x4. Unhandled bits: [5] when writing value 0x20. Tags: MMS (0x2).
08:46:53 [WARNING] timer3: Unhandled write to offset 0x18. Unhandled bits: [5-6] when writing value 0x60. Tags: OC1M (0x6).
08:46:53 [WARNING] timer3: Unhandled write to offset 0x18. Unhandled bits: [3] when writing value 0x8. Tags: OC1PE (0x1).
08:46:53 [WARNING] gpioPortB: Unhandled write to offset 0x4. Unhandled bits: [4] when writing value 0x10. Tags: OT4 (0x1).
08:46:53 [WARNING] gpioPortB: Unhandled write to offset 0x20. Unhandled bits: [16] when writing value 0x10000. Tags: AFSEL4 (0x1).
08:46:53 [WARNING] gpioPortB: Unhandled write to offset 0x24. Unhandled bits: [5] when writing value 0x20. Tags: AFSEL9 (0x2).
08:46:53 [WARNING] timer17: Unhandled write to offset 0x18. Unhandled bits: [0] when writing value 0x1. Tags: CC1S (0x1).
08:46:53 [WARNING] gpioPortC: Unhandled write to offset 0x20. Unhandled bits: [16] when writing value 0x10000. Tags: AFSEL4 (0x1).
08:46:53 [WARNING] gpioPortC: Unhandled write to offset 0x20. Unhandled bits: [20] when writing value 0x100000. Tags: AFSEL5 (0x1).
08:46:53 [WARNING] timer2: Unhandled write to offset 0xC. Unhandled bits: [11] when writing value 0x800. Tags: Capture/Compare 3 DMA request enable (CC3DE) (0x1).
08:46:53 [WARNING] timer2: Unhandled write to offset 0x20. Unhandled bits: [8] when writing value 0x100. Tags: CC3E (0x1).
08:46:53 [WARNING] timer3: Unhandled write to offset 0xC. Unhandled bits: [9] when writing value 0x200. Tags: Capture/Compare 1 DMA request enable (CC1DE) (0x1).
08:46:53 [WARNING] timer3: Unhandled write to offset 0x20. Unhandled bits: [0] when writing value 0x1. Tags: CC1E (0x1).
08:46:53 [WARNING] usart2: Unhandled write to offset 0x8. Unhandled bits: [0] when writing value 0x1. Tags: EIE (0x1).
08:46:53 [WARNING] usart2: Unhandled write to offset 0x8. Unhandled bits: [6] when writing value 0x40. Tags: DMAR (0x1).
08:47:05 [WARNING] usart3: Char was received, but the receiver (or the whole USART) is not enabled. Ignoring.
08:47:05 [WARNING] usart3: Char was received, but the receiver (or the whole USART) is not enabled. Ignoring.
08:47:06 [WARNING] usart3: Char was received, but the receiver (or the whole USART) is not enabled. Ignoring.
08:47:06 [WARNING] usart3: Char was received, but the receiver (or the whole USART) is not enabled. Ignoring.
08:47:07 [WARNING] usart3: Char was received, but the receiver (or the whole USART) is not enabled. Ignoring.
08:47:07 [WARNING] usart3: Char was received, but the receiver (or the whole USART) is not enabled. Ignoring.
08:47:07 [WARNING] usart3: Char was received, but the receiver (or the whole USART) is not enabled. Ignoring.
08:47:08 [WARNING] usart3: Char was received, but the receiver (or the whole USART) is not enabled. Ignoring.

It is caused by using WithTaggedFlag or WithTag for registers used in startup phase.

For Timers this warnings appears when writing to CaptureOrCompareMode1/2 registers - these flags are not defined because there is different tagging depending on timer configuration(input/output)

Same for USART

PiotrZierhoffer, 2021-08-17:
This is referenced in https://github.com/renode/renode-infrastructure/pull/52

PiotrZierhoffer, 2022-02-01:
Closing as the warnings were mostly there "by design" (we don't implement the whole logic of the peripheral).

closed 2022-02-01 by PiotrZierhoffer