renode/renode #237: stm32f407 ethernet mac does not works properly with zephyr

katyo, 2021-08-12:
I'm trying run telnet daemon example from zephyr which build for stm32f4.
My renode script looks like following:
```
:name: Run on STM32F4
:description: This script runs ssh daemon on STM32F4 Discovery.

$name?="STM32F4_Discovery"
$bin?=@build/zephyr/zephyr.elf
$plat?=@platforms/boards/stm32f4_discovery-kit.repl
$tap?="tap0"

using sysbus

emulation CreateServerSocketTerminal 3456 "console"
emulation CreateSwitch "switch"
emulation CreateTap $tap "tap" true
connector Connect host.tap switch

mach create $name
machine LoadPlatformDescription $plat
cpu PerformanceInMips 125

logLevel 3 nvic

connector Connect sysbus.uart2 console
connector Connect sysbus.ethernet switch

macro reset
"""
    sysbus LoadELF $bin
"""

runMacro $reset
```

I add appropriate IP to `tap0` and setup routing rules. Next I `start` program and try connect to telnet but it fails:
```
Trying 192.0.2.1...
telnet: Unable to connect to remote host: No route to host
```

My renode console output is following:
```
16:18:31.9727 [WARNING] ethernet.phy: Unhandled write to offset 0x0, value 0x8000.
16:18:33.1809 [WARNING] ethernet.phy: Unhandled write to offset 0x0, value 0x1000.
16:18:33.2001 [WARNING] ethernet: Unhandled write to offset 0x8, value 0x0.
16:18:33.2002 [WARNING] ethernet: Unhandled write to offset 0xC, value 0x0.
16:18:33.2091 [WARNING] ethernet: Unhandled write to offset 0x1C, value 0x0.
16:18:33.2095 [WARNING] ethernet: Unhandled read from offset 0x1C.
16:18:33.2166 [WARNING] ethernet: Unhandled write to offset 0x1C, value 0x0.
16:18:33.2421 [INFO] ethernet: Setting RDLA to 0x20003328.
16:18:33.2677 [WARNING] uart2: Unhandled write to offset 0x0. Unhandled bits: [0-2, 4, 8-31] when writing value 0xFFFFFFBF. Tags: PE (0x1), FE (0x1), NF (0x1), IDLE (0x1), LBD (0x1), CTS (0x1), RESERVED (0x3FFFFF). (67)
16:46:02.8557 [ERROR] ethernet: Dropping - no packets sent. (62)
```
The last message repeats when I try connect.

Meanwhile Zephyr properly configures network and runs telnet daemon:
```
*** Booting Zephyr OS build zephyr-v2.6.0-1923-g72bb75a360ce  ***

[00:00:00.263,000] <inf> shell_telnet: Telnet shell backend initialized
[00:00:00.263,000] <inf> net_config: Initializing network
[00:00:00.263,000] <inf> net_config: Waiting interface 1 (0x20000b20) to be up...
[00:00:00.761,000] <inf> net_config: Interface 1 (0x20000b20) coming up
[00:00:00.761,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.761,000] <inf> net_sshd: Starting Telnet sample
[00:00:00.761,000] <inf> net_sshd: IPv4 address: 192.0.2.1
```

If I understand right function `SendFrames` expected to be called to do actual transfer but it does not happens (https://github.com/renode/renode-infrastructure/blob/c6f3af35c129d58e1e20779a81521663878f1940/src/Emulator/Peripherals/Peripherals/Network/SynopsysEthernetMAC.cs#L145).
Zephyr's driver implementation located here (https://github.com/zephyrproject-rtos/zephyr/blob/1b238b8bd39fdec5fb3339fc4c08ddc8233abc84/drivers/ethernet/eth_stm32_hal.c)

mateusz-holenko, 2021-08-17:
Hi @katyo, thanks for reporting the issue!

Indeed https://github.com/renode/renode-infrastructure/blob/c6f3af35c129d58e1e20779a81521663878f1940/src/Emulator/Peripherals/Peripherals/Network/SynopsysEthernetMAC.cs#L198 looks quite suspicious - it looks that the model doesn't accept any incoming frames until the first one is sent.

I'm not sure why is it so - I'll take a look at that.

In the meantime you can set this flag to `true` manually in Monitor before starting the simulation:
```
# this will allow to access and modify private model members
(machine-0) allowPrivates true
(machine-0) ethernet packetSent true
```

Could you check if that solves the problem for you?

katyo, 2021-08-17:
@mateusz-holenko It works. I successfully connected to telnetd from host.

sergey-perepelkin, 2025-08-05:
Hi @katyo 
Could you please share overlays for building FW for stm32f4?