renode/renode #238: Can't run the test firmware on stm32f103.repl

miguelsoares14, 2021-08-12:
Hi guys, I'm new at Renode, however I found it a very interesting platform and so I'm trying to do some tests to gain some experience with it.
For that, I decided to develop some simple code for cpu STM32F103C8 (also known as BluePill), for that I used PlantformIO with arduino framework. After compiling it, I uploaded the .elf file to my renode machine I set it to start, however I'm cathing this error:
[image]
Apparently its a problem when accessing the RCC CR register for clock configuration. On stm32f103.repl file there a Tag with: 
        Tag <0x40021000, 0x40021003> "RCC_CR" 0x0A020083
On SVD file there no register or peripheral named with "RCC_CR", I tried to change it to "CR" or "RCC", also the Tag is writing a 4 bytes registers into a space of 3 bytes (0x40021003 - 0x40021000) which is also strange... However the problem remains. 
Can anyone help me?

PiotrZierhoffer, 2021-08-17:
This is actually 4 bytes, the range is inclusive.

The value was prepared for some specific software and you might use another one. I'd try

 ```
Tag <0x40021000, 0x40021003> "RCC_CR" 0xFFFFFFFF
```

Or just use a value that you know your software is expecting. You can see where you got stuck with `cpu LogFunctionNames true`

PiotrZierhoffer, 2021-09-21:
Let's close this for now, please reopen if you have any other comments or questions

closed 2021-09-21 by PiotrZierhoffer