renode/renode #242: Binary Translation Working Principle

felix-boeseler, 2021-09-07:
Hello,

How does renode realize the translation/execution of guest code on the host machine exactly. For instance, OVPsim (https://www.ovpworld.org/) uses Dynamic Binary Translation (DBT) on basic blocks. For this, debug accesses are used to prefetch all instructions of the basic block from the memory without side effects and timings. The concept of debug accesses is also existent in SystemC. However, I could not find these debug accesses in renode. For the peripheral interface classes I can only implement read and write methods. Have I overseen something? Does renode not work with DBT (e.g., an interpretive approach)?
Many regards and many thanks in advance

PiotrZierhoffer, 2021-09-09:
Renode does rely on binary translation, that's correct. For execution we use data from `MappedMemory` - this is connected to our translation libraries via raw C-style pointers. In this case the memory is acting as a simple data array, and there are no side effects or timing issues. Also see my answer in #244 as this might explain our approach to modelling.
  
  If there are multiple ways to talk to a memory peripheral (e.g. a flash controller with XIP) we tend to implement both the C# R/W interface and the `MappedMemory` interface for execution.

felix-boeseler, 2021-09-09:
Thanks for the detailed and fast answer!

When I read _“read/write operations executed by the CPU (usually in the C implementation in the tlib submodule) are either directed to the internal memory or passed to the system bus and handled by the framework at the C# level”_ [1] I thought for some reason that _internal memory_ means only things like memory mapped status registers of a CPU. It then also makes sense how the dynamic binary translation can be implemented efficiently. 

So, the executable must be placed in a `MappedMemory ` area and must not be implemented with the C# R/W interfaces? However, for non-executable code I may define own C# R/W interface-based memory components (e.g., with custom side effects internally)?

Furthermore, a peripheral is **either** `MappedMemory` or implemented with the C# R/W interfaces in the eyes of one CPU. The approach is **not** to prefetch instruction with the `MappedMemory ` interface and then execute the real memory accesses later with the C# R/W interfaces. Is this correct?

[1] https://renode.readthedocs.io/en/latest/advanced/writing-peripherals.html