renode/renode #243: STM32 RCC peripheral

kaizoku-oh, 2021-09-07:
Is it possible to use  STM32 with external oscillator if so Is there any example that uses STM32 with external oscillator?

PiotrZierhoffer, 2021-09-09:
In general, Renode does not simulate clock sources. While we do have an RCC peripheral, it's a bit of a mock - we try to make the software "believe" it uses the real thing, but it doesn't really change much in the emulation.
Please see a comment in our STM32F4 RCC model: https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/Miscellaneous/STM32F4_RCC.cs#L22

If it doesn't work for you then you might need to improve this model a bit - but this should be very simple, as the stub should guide you the proper way.

You might also need to modify the REPL file with proper frequencies of timers etc, if you need it.

The first thing I'd try is "just run and see what happens".

PiotrZierhoffer, 2021-12-29:
I'll close this issue but feel free to reopen or file a new one if you have additional questions

closed 2021-12-29 by PiotrZierhoffer