renode/renode #245: s @scripts/single-node/stm32f4_discovery.resc - there are some warning

phuongltbk, 2021-09-09:
Hello Renode team,

I am new member to use Renode
I am following guideline document in https://renode.readthedocs.io/en/latest/introduction/demo.html
I am using version 1.12.0.37823
After I run command
s @scripts/single-node/stm32f4_discovery.resc

I got some warning 
08:11:48.8032 [WARNING] rcc: Unhandled write to offset 0xC, value 0x0.
08:11:48.8064 [WARNING] sysbus: Read from an unimplemented register PWR:CR (0x40007000), returning a value from SVD: 0x0.
08:11:48.8071 [WARNING] sysbus: Write of value 0x4000 to an unimplemented register PWR:CR (0x40007000) generated from SVD.
08:11:48.8101 [WARNING] rcc: Unhandled write to offset 0x4. Unhandled bits: [29] when writing value 0x7405408. Tags: RESERVED (0x2).
08:11:48.8104 [WARNING] sysbus: Write of value 0x605 to an unimplemented register FLASH:ACR (0x40023C00) generated from SVD.
08:11:48.8215 [WARNING] gpioPortC: Unhandled write to offset 0x24. Unhandled bits: [11] when writing value 0x800. Tags: AFSEL10 (0x8).
08:11:48.8215 [WARNING] gpioPortC: Unhandled write to offset 0x24. Unhandled bits: [15] when writing value 0x8000. Tags: AFSEL11 (0x8).
08:11:48.8231 [WARNING] uart4: Unhandled read from offset 0x14. (2)
08:11:48.8235 [WARNING] uart4: Unhandled write to offset 0x14, value 0x0.
08:11:48.8315 [WARNING] timer7: [cpu: 0x8001C00] Attempt to read Word from peripheral that doesn't support Word interface. Offset 0x0.
08:11:48.8318 [WARNING] timer7: [cpu: 0x8001C34] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x0, value 0x0.
08:11:48.8320 [WARNING] timer7: [cpu: 0x8001C34] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x28, value 0x0.
08:11:48.8321 [WARNING] timer7: [cpu: 0x8001C68] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x14, value 0x1.
08:11:48.8322 [WARNING] timer7: [cpu: 0x8001CA0] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x28, value 0x347.
08:11:48.8323 [WARNING] timer7: [cpu: 0x8001CA0] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x14, value 0x1.
08:11:48.8324 [WARNING] timer7: [cpu: 0x8001D04] Attempt to read Word from peripheral that doesn't support Word interface. Offset 0x0.
08:11:48.8325 [WARNING] timer7: [cpu: 0x8001D08] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x0, value 0x0.
08:11:48.8325 [WARNING] timer7: [cpu: 0x8001D6C] Attempt to read Word from peripheral that doesn't support Word interface. Offset 0x20.
08:11:48.8326 [WARNING] timer7: [cpu: 0x8001D6C] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x20, value 0x0.
08:11:48.8327 [WARNING] timer7: [cpu: 0x8001D6C] Attempt to read Word from peripheral that doesn't support Word interface. Offset 0x20.
08:11:48.8328 [WARNING] timer7: [cpu: 0x8001D6C] Attempt to read Word from peripheral that doesn't support Word interface. Offset 0x4.
08:11:48.8328 [WARNING] timer7: [cpu: 0x8001D6C] Attempt to read Word from peripheral that doesn't support Word interface. Offset 0x18.
08:11:48.8329 [WARNING] timer7: [cpu: 0x8001DF2] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x4, value 0x0.
08:11:48.8330 [WARNING] timer7: [cpu: 0x8001DF2] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x18, value 0x0.
08:11:48.8334 [WARNING] timer7: [cpu: 0x8001DF2] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x20, value 0x1.
08:11:48.8335 [WARNING] timer7: [cpu: 0x8002070] Attempt to read Word from peripheral that doesn't support Word interface. Offset 0x18.
08:11:48.8335 [WARNING] timer7: [cpu: 0x8002070] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x18, value 0x0.
08:11:48.8336 [WARNING] timer7: [cpu: 0x80025FC] Attempt to read Word from peripheral that doesn't support Word interface. Offset 0xC.
08:11:48.8337 [WARNING] timer7: [cpu: 0x800260A] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0xC, value 0x1.
08:11:48.8338 [WARNING] timer7: [cpu: 0x8001D50] Attempt to read Word from peripheral that doesn't support Word interface. Offset 0x0.
08:11:48.8338 [WARNING] timer7: [cpu: 0x8001D60] Attempt to write Word from peripheral that doesn't support Word interface. Offset 0x0, value 0x1.
08:11:48.8353 [WARNING] sysbus: Read from an unimplemented register SYSCFG:EXTICR3 (0x40013810), returning a value from SVD: 0x0.
08:11:48.8355 [WARNING] sysbus: Write of value 0x0 to an unimplemented register SYSCFG:EXTICR3 (0x40013810) generated from SVD.
08:11:48.8357 [WARNING] sysbus: Read from an unimplemented register SYSCFG:EXTICR3 (0x40013810), returning a value from SVD: 0x0.
08:11:48.8358 [WARNING] sysbus: Write of value 0x0 to an unimplemented register SYSCFG:EXTICR3 (0x40013810) generated from SVD.
08:11:48.8368 [WARNING] sysbus: Read from an unimplemented register SYSCFG:EXTICR1 (0x40013808), returning a value from SVD: 0x0.
08:11:48.8370 [WARNING] sysbus: Write of value 0x0 to an unimplemented register SYSCFG:EXTICR1 (0x40013808) generated from SVD.
08:11:48.8371 [WARNING] sysbus: Read from an unimplemented register SYSCFG:EXTICR1 (0x40013808), returning a value from SVD: 0x0.
08:11:48.8372 [WARNING] sysbus: Write of value 0x0 to an unimplemented register SYSCFG:EXTICR1 (0x40013808) generated from SVD.
08:11:48.8437 [WARNING] sysbus: Read from an unimplemented register SYSCFG:PMC (0x40013804), returning a value from SVD: 0x0.
08:11:48.8438 [WARNING] sysbus: Write of value 0x800000 to an unimplemented register SYSCFG:PMC (0x40013804) generated from SVD.
08:11:48.8439 [WARNING] gpioPortA: Unhandled write to offset 0x20. Unhandled bits: [4-5, 7] when writing value 0xB0. Tags: AFSEL1 (0xB).
08:11:48.8440 [WARNING] gpioPortA: Unhandled write to offset 0x20. Unhandled bits: [8-9, 11] when writing value 0xB00. Tags: AFSEL2 (0xB).
08:11:48.8441 [WARNING] gpioPortA: Unhandled write to offset 0x20. Unhandled bits: [28-29, 31] when writing value 0xB0000000. Tags: AFSEL7 (0xB).
08:11:48.8442 [WARNING] gpioPortB: Unhandled write to offset 0x24. Unhandled bits: [8-9, 11] when writing value 0xB00. Tags: AFSEL10 (0xB).
08:11:48.8443 [WARNING] gpioPortB: Unhandled write to offset 0x24. Unhandled bits: [12-13, 15] when writing value 0xB000. Tags: AFSEL11 (0xB).
08:11:48.8443 [WARNING] gpioPortB: Unhandled write to offset 0x24. Unhandled bits: [16-17, 19] when writing value 0xB0000. Tags: AFSEL12 (0xB).
08:11:48.8444 [WARNING] gpioPortB: Unhandled write to offset 0x24. Unhandled bits: [20-21, 23] when writing value 0xB00000. Tags: AFSEL13 (0xB).
08:11:48.8445 [WARNING] gpioPortC: Unhandled write to offset 0x20. Unhandled bits: [4-5, 7] when writing value 0xB0. Tags: AFSEL1 (0xB).
08:11:48.8446 [WARNING] gpioPortC: Unhandled write to offset 0x20. Unhandled bits: [16-17, 19] when writing value 0xB0000. Tags: AFSEL4 (0xB).
08:11:48.8447 [WARNING] gpioPortC: Unhandled write to offset 0x20. Unhandled bits: [20-21, 23] when writing value 0xB00000. Tags: AFSEL5 (0xB).
08:11:48.8647 [WARNING] ethernet.phy: Unhandled write to offset 0x0, value 0x8000.
08:11:49.0329 [WARNING] ethernet.phy: Unhandled write to offset 0x0, value 0x1000.
08:11:49.0331 [WARNING] ethernet: Unhandled write to offset 0x8, value 0x0.
08:11:49.0332 [WARNING] ethernet: Unhandled write to offset 0xC, value 0x0.
08:11:49.0332 [WARNING] ethernet: Unhandled write to offset 0x1C, value 0x0.

Please help me how to solve it

Best regards,
Phuong Le

PiotrZierhoffer, 2021-09-14:
Hi @phuongltbk 

We usually don't implement full models of peripherals - only the bits we need. So if you see these warnings it means there are some reads and writes to models that we don't handle.

It's normal and very often it doesn't hurt. It really depends on the application you're trying to run.

One issue I see in your logs is unhandled word access to timer7. Frankly, I can see we have a fix already published: https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/Timers/STM32_Timer.cs#L18

I would recommend you try and use the latest sources and see if the issue persists.

PiotrZierhoffer, 2021-12-29:
I'll close this issue but feel free to reopen or file a new one if you have additional questions

closed 2021-12-29 by PiotrZierhoffer