renode/renode #250: emulation's load never less than 1.0

quangr, 2021-09-27:
Hi,Renode's developers.
 I find that when i reduce the cpu's PerformanceInMips, the machine's load will become very low, but the emulation's laod will always greater than 1.0.
[image]

 I look up in source code and find that the SlaveTimeSource will sleep in InnerExecute function before report, which will make slave excute time no less than grant time.

the following codes are at SlaveTimeSource.cs:234
```
            while(waitingForSlave || (timeLeft >= quantum && isStarted))
            {
                waitingForSlave = false;
                var syncPointReached = InnerExecute(out var elapsed, slave:true); // HERE WILL SLEEP!
                timeLeft -= elapsed;
                if(!syncPointReached)
                {
                    // we should not ask for time grant since the current one is not finished yet
                    waitingForSlave = true;
                    TimeHandle.ReportBackAndBreak(timeLeft);
                    return true;
                }
            }

            TimeHandle.ReportBackAndContinue(timeLeft);
```
 I don't know if there's something i don't understand about this behavior but when i remove the slave's sleep time the emulation's load go normal, so i was thinking maybe the slave should not sleep since all it have to do is finish its work in grant time.

BTW, thanks for developing such a cool software so i can test my firmware at my pc!

PiotrZierhoffer, 2021-09-29:
Hi @quangr

Thanks for your kind words!

The sleep is to simulate real time flow, but it's possible to disable it.

`emulation SetGlobalAdvanceImmediatelly true` should yield similar results.

The load is above 1 because when we actually have a lot of computation going on, we're a bit slower than real time, but we catch up during sleeping - but never faster than real time: that causes the average to be over 1. With the setting I mention above we skip the virtual time to the next scheduled event (e.g. a timer tick).

quangr, 2021-09-29:
Thanks for your detailed and fast answer! And i have a pull requests about a tlib error where arm qsub16 instruction will be executed as qadd16, please check it: https://github.com/antmicro/tlib/pull/9. thanks again for your replying

PiotrZierhoffer, 2021-10-12:
Hi @quangr 

Actually it did not get merged because we're working on supporting nRF52840 tests in Zephyr, and this is already fixed on our internal head. We should release it very soon.

PiotrZierhoffer, 2022-02-01:
I believe this issue can be closed. Feel free to re-open if you need any further input

closed 2022-02-01 by PiotrZierhoffer