renode/renode #251: Scripts fail to download binaries (reason: SecureChannelFailure)

bhjortsberg, 2021-09-30:
Trying renode for the first time and I'm getting download errors when trying to run any script (I've tried multiple)

```
(monitor) s @scripts/single-node/stm32f4_discovery.resc
Failed to download https://dl.antmicro.com/projects/renode/svd/STM32F40x.svd.gz, see log for details.
Failed to download https://dl.antmicro.com/projects/renode/stm32f4discovery.elf-s_445441-827a0dedd3790f4559d7518320006613768b5e72, see log for details.
(STM32F4_Discovery) lastLog 
10:24:57.2744 [INFO] Downloading https://dl.antmicro.com/projects/renode/svd/STM32F40x.svd.gz. 
10:24:57.3211 [ERROR] Failed to download from https://dl.antmicro.com/projects/renode/svd/STM32F40x.svd.gz, reason: SecureChannelFailure (attempt 2/5) 
...

```

PiotrZierhoffer, 2021-09-30:
Hi @bhjortsberg 

Can you please provide some details on the machine you're using? OS and Mono version would be helpful. Is it a regular machine or a container?

bhjortsberg, 2021-10-01:
> Hi @bhjortsberg
> 
> Can you please provide some details on the machine you're using? OS and Mono version would be helpful. Is it a regular machine or a container?

Hi,
I'm running Renode on Windows 10 on my laptop.

Badboy1307, 2021-10-08:
Same props here as well I'm also using Windows 10 laptop. Any solution for this?

DanielHabenicht, 2021-10-31:
When I tried running the newest nightly build (`renode_1.12.0+20211030git42c40f85.msi`) I ran into this error: 

```bash
PS C:\Users\danie> renode
Fatal error:
The type initializer for 'Antmicro.Renode.Core.EmulationManager' threw an exception.
   at Antmicro.Renode.Core.EmulationManager.get_Instance()
   at Antmicro.Renode.UI.CommandLineInterface.Run(Options options, Action`1 beforeRun)
   at Antmicro.Renode.Program.<>c__DisplayClass0_0.<Main>b__1()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()
Inner exception:
Could not load file or assembly 'IronPython, Version=2.7.11.0, Culture=neutral, PublicKeyToken=7f709c5b713576e1' or one of its dependencies. The system cannot find the file specified.
   at Antmicro.Renode.Core.EmulationManager..ctor()
   at Antmicro.Renode.Core.EmulationManager.RebuildInstance()
   at Antmicro.Renode.Core.EmulationManager..cctor()

Unhandled Exception: System.TypeInitializationException: The type initializer for 'Antmicro.Renode.Core.EmulationManager' threw an exception. ---> System.IO.FileNotFoundException: Could not load file or assembly 'IronPython, Version=2.7.11.0, Culture=neutral, PublicKeyToken=7f709c5b713576e1' or one of its dependencies. The system cannot find the file specified.
   at Antmicro.Renode.Core.EmulationManager..ctor()
   at Antmicro.Renode.Core.EmulationManager.RebuildInstance()
   at Antmicro.Renode.Core.EmulationManager..cctor()
   --- End of inner exception stack trace ---
   at Antmicro.Renode.Core.EmulationManager.get_Instance()
   at Antmicro.Renode.UI.CommandLineInterface.Run(Options options, Action`1 beforeRun)
   at Antmicro.Renode.Program.<>c__DisplayClass0_0.<Main>b__1()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()
```

But as a workaround `renode_1.12.0+20210908git4c843c14.msi` seems to work.

PiotrZierhoffer, 2021-11-03:
This seems to be unrelated, but we'll check it shortly. Thanks for reporting this!

g40, 2024-04-29:
This kind of thing makes for a terrible user experience. I'm running a completely new install `.MSI` from Github on Windows 10x64 22H2 and it fails in exactly the same (unexplained) manner. 

Any thoughts?

FWIW `t41.resc` is simply a renamed STM32F7 file.

```
renode -p ./t41.resc
17:36:38.0278 [INFO] Loaded monitor commands from: R:\apps\renode\./scripts/monitor.py
17:36:38.8796 [INFO] Including script: R:\src\embedded\teensy\gdbt4x\t41.resc
17:36:38.9006 [INFO] System bus created.
17:36:39.4946 [DEBUG] Segment size automatically calculated to value 16MiB
17:36:39.4966 [DEBUG] Segment size automatically calculated to value 64KiB
17:36:39.4966 [DEBUG] Segment size automatically calculated to value 128KiB
17:36:39.6516 [DEBUG] Segment size automatically calculated to value 64KiB
17:36:39.7776 [INFO] Downloading http://antmicro.com/projects/renode/svd/STM32F7x6.svd.gz.
17:36:39.9051 [ERROR] Failed to download from http://antmicro.com/projects/renode/svd/STM32F7x6.svd.gz, reason: SecureChannelFailure.
```