renode/renode #258: stm32f7_discovery-bb.repl : no output to usart1 Analyzer

rpcme, 2021-10-31:
Hello,

I am trying to get a UART demo running using the stm32f7_discovery-bb.repl platform.
I have verified that output displays on the physical board when connecting to /dev/ttyACM0 on the ST-Link port.

My renode version is: Renode, version 1.12.0.37823 (44d6786a-202104022100)
I have tried running on Ubuntu 18.04 and 20.04.

Are there some special settings I need to do to make it work?  How it's setup (it's super basic):

```c
  xUARTHandle.Init.BaudRate = 115200;
  xUARTHandle.Init.WordLength = UART_WORDLENGTH_8B;
  xUARTHandle.Init.StopBits = UART_STOPBITS_1;
  xUARTHandle.Init.Parity = UART_PARITY_NONE;
  xUARTHandle.Init.HwFlowCtl = UART_HWCONTROL_NONE;
  xUARTHandle.Init.Mode = UART_MODE_RX | UART_MODE_TX;
  BSP_COM_Init( COM1, &xUARTHandle );
```

And just using the blocking call `HAL_UART_Transmit`.

I am seeing some noise on the log:

```text
02:00:25.0128 [INFO] STM32F746: Machine started.
02:00:25.1246 [WARNING] nvic: Unhandled write to offset 0xF50, value 0x0.
02:00:25.1246 [WARNING] nvic: Unhandled write to offset 0xD14, value 0x30000.
02:00:25.1247 [WARNING] nvic: Unhandled write to offset 0xD84, value 0x0.
02:00:25.1247 [WARNING] nvic: Unhandled read from offset 0xD80.
02:00:25.1247 [WARNING] nvic: Unhandled write to offset 0xF60, value 0x0.
02:00:25.1248 [WARNING] nvic: Unhandled write to offset 0xD14, value 0x10000.
02:00:25.1799 [WARNING] gpioPortA: Unhandled write to offset 0x24. Unhandled bits: [4-6] when writing value 0x70. Tags: AFSEL9 (0x7).
02:00:25.1811 [WARNING] gpioPortB: Unhandled write to offset 0x20. Unhandled bits: [28-30] when writing value 0x70000000. Tags: AFSEL7 (0x7).
```

Please let me know if there's some special pin setting I need to do to talk to this UART - I kinda hope not because I would like to use the same between renode and the physical board.

Thank you

PiotrZierhoffer, 2021-11-09:
Hi @rpcme 

Renode does not need any special settings for pins, the software should be the same as on hardware.

We'd need more information to help you debug it. 
You can get better logs if you run the following in the Renode's Monitor:
```
sysbus LogAllPeripheralsAccess true
cpu LogFunctionNames true
```

This will show us if there's any interaction with the UART at all.
If you can share the ELF file and the resc script you're using, please attach it here so we'll try to take a look.

rpcme, 2021-11-09:
Hello @PiotrZierhoffer - attached are the files.

[sample.tar.gz](https://github.com/renode/renode/files/7504840/sample.tar.gz)

The debug settings are amazing - I am not sure how I missed that in the documentation.  I do see that the errors are happening right after calling board initialization routines like ```SCB_EnableICache();```. 

When I run against the device, I receive this from UART.

```text
rpcme@proteus:~/src/freertos-stm32f746/stm32f746-parts/uart$ picocom -b 115200 /dev/ttyACM0 
picocom v3.1

port is        : /dev/ttyACM0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready
Solid
HAL_OK
Solid
HAL_OK
```
and so on ...

PiotrZierhoffer, 2022-02-01:
Hi @rpcme . The fix is on master, you should be able to verify it with the latest nightly build from https://dl.antmicro.com/projects/renode/builds/

I can see the output like this:
```
Solid
HAL_OK
```

closed 2022-02-01 by PiotrZierhoffer