renode/renode #259: [STMCAN] EXIDMASK is missing some bits

sanko-oz, 2021-11-02:
It seams like the EXIDMASK is missing 4 bits, which leads to incorrect CAN identifiers. 
The code in questions is:

```
public const int STIDSHIFT = 21;
public const uint STIDMASK = 0x7FF;
public const int EXIDSHIFT = 3;
public const uint EXIDMASK = 0x3FFF;
public const int IDESHIFT = 2;
public const uint IDEMASK = 0x1;
public const int RTRSHIFT = 1;
public const uint RTRMASK = 0x1;
```
Doing following will result in a gap of 4 bits between STIDMASK and EXIDMASK :
```
(STIDMASK << STIDSHIFT ) + (EXIDMASK << EXIDSHIFT ) + (IDEMASK << IDESHIFT ) + (RTRMASK  << RTRSHIFT )
=
(0x7FF<< 21) + (0x3FFF<< 3) + (0x1<< 2) + (0x1<< 1)
```

The problem is EXIDMASK, which should be 0x3FFF **F**  because the extended identifier is 18 bits long not 14.
We rebuild renode with that change and it fixes the issue.

mateusz-holenko, 2021-11-09:
Hi @sanko-oz , thanks for reporting the issue.

Indeed it looks like a bug, we'll address it shortly.

mateusz-holenko, 2021-11-10:
075803e4a3e2259d0a57d17e3bb8ca849911b828 should fix the problem

closed 2021-11-15 by mateusz-holenko