renode/renode #260: LogToWireshark on Windows uses wrong pipe

agudek, 2021-11-03:
I am trying to capture the simulation traffic in Windows. 
Using command `emulation LogToWireshark wireless` or `emulation LogWirelessTraffic` gives the following error in Wireshark :

_The capture session on "\\.\pipe\wireshark-allWirelessTraffic" could not be started due to error on pipe open: The system cannot find the file specified.  (2)._ 

Checking all open pipes on Windows with `[System.IO.Directory]::GetFiles("\\.\\pipe\\")` shows a pipe with the name \\\\.\pipe\pipe\wireshark-allWirelessTraffic instead of the expected \\\\.\pipe\wireshark-allWirelessTraffic, and I can correctly connect to the pipe using Wireshark by running `Wireshark.exe -i \\.\pipe\pipe\wireshark-allWirelessTraffic -k`

Is there way of easily changing the name of the pipe created or changing the default behaviour when the emulation command is run?

PiotrZierhoffer, 2021-11-09:
Hi @agudek, thanks for the report.

We have a fix for that pending to be released. Hope to be able to push it soon!

mateusz-holenko, 2021-12-15:
This should be fixed with: https://github.com/renode/renode/commit/fb16507c8dc5013822943fa7309b141873c56a7f.

Could you check if it works for you now, @agudek?

agudek, 2021-12-15:
Yes, that fixes it. 👍

closed 2021-12-15 by agudek